Ultimaker/Cura #6738: [4.4.0] Custom Profiles Resetting

TKDesigns, 2019-12-01:
When starting Cura 4.4.0, it opens up under my custom PLA profile. I open a saved file and it loads the custom PLA profile with whatever changes I had made for that specific print, and it shows the star as it's not a saved profile. When I click on the original custom PLA profile it loads but is an older saved profile. I've added some screenshots to show what I'm talking about and highlighted some of the items that where changed.
[image]
[image]
[image]
[image]
[image]

nallath, 2019-12-02:
Could you share the project file? That makes it a lot easier for us to reproduce

TKDesigns, 2019-12-02:
Yes, I can when I get home tonight!

TKDesigns, 2019-12-02:
I can't seem to attach the file. It's an unsupported file format. But It shouldn't really matter as long as the file you open was saved with a custom profile that has had a few items changes from the original.

Ghostkeeper, 2019-12-05:
You can .zip it up and attach that to Github.

Ghostkeeper, 2019-12-05:
I think what happened here is that the project file was saved earlier with your older version of your custom profile. When loading a project file, Cura will see that the project file contains a profile called PLA Standard Quality, but you also have a locally installed profile called PLA Standard Quality. So in a dialogue it asks you whether you want to override the profile or create a new one. If you selected to override (the default I think) then it will remove your more recent changes and use the profile in the project file.

TKDesigns, 2019-12-05:
> I think what happened here is that the project file was saved earlier with your older version of your custom profile. When loading a project file, Cura will see that the project file contains a [...]

Yes, I think this is correct, but it shouldn't really do that with a standard saved profile. It loads the profile it used with any changes and shows it with a star. So it shouldn't change a standard profile. The updated existing option is what I chose, and the create new adds a second saved custom profile in my profile list. I can't be doing that with every saved file, I'll wind up with 50 custom profiles. Lol

TKDesigns, 2019-12-05:
If you save a file and wind up changing the custom profile used when you save it, it should just open up that file with the profile opening up with a star. The only way around this is to go though and re-save every file I have with the updated custom profile every time I change it. And that is extremely cumbersome and just not feasible.

Ghostkeeper, 2019-12-06:
> So it shouldn't change a standard profile.

Indeed, they are read-only and also protected by your operating system sometimes. Cura can't change them even if it wanted to. In your case though your "PLA Standard Quality" is not a built-in profile but one that was added by yourself, right?

TKDesigns, 2019-12-07:
> > So it shouldn't change a standard profile.
> 
> Indeed, they are read-only and also protected by your operating system sometimes. Cura can't change them even if it wanted to. In your case though your "PLA Standard Quality" is not a built-in profile but one that was added by yourself, right?

Correct. It is a custom profile added by me. But those really shouldn't change either unless it's resaved by me.

Ghostkeeper, 2019-12-09:
Then I suggest that you select the option to create a new profile instead, in the profile loading pop-up.

Can I ask what exactly you're expecting to see and what are you seeing instead? It's in our bug reporting template.

TKDesigns, 2019-12-09:
> Then I suggest that you select the option to create a new profile instead, in the profile loading pop-up.
> 
> Can I ask what exactly you're expecting to see and what are you seeing instead? It's in our bug reporting template.

I don't want to create a new profile because every time I open a saved project it will create a new custom profile and I really don't want to end up with 50 pla profiles.

Basically, when I save a project and I've used a slightly modified custom profile, it shouldn't be saving the stock custom profile in the background and over righting it from any updates the next time I open it. If that makes sense. While I can change a custom profile unlike the stock profiles, they should be treated the same when saving a project and not come up with different settings.

Ghostkeeper, 2019-12-10:
You don't want the old profile to change, but you don't want it to create a new profile either? So you want to load just the model and not the settings?

TKDesigns, 2019-12-10:
> You don't want the old profile to change, but you don't want it to create a new profile either? So you want to load just the model and not the settings?

Sort of. When I save a project, it should save all the settings that were made for that particular project. But it shouldn't change "factory" settings, whether cura's profiles, or mine.

Ghostkeeper, 2019-12-11:
> Sort of. When I save a project, it should save all the settings that were made for that particular project. But it shouldn't change "factory" settings, whether cura's profiles, or mine.

When opening a project Cura should ask whether to load just the model or also the settings. Unless you changed that in your application preferences (you can choose there to always only load the model or always load the settings as well, or ask every time).

> When I change these things, the profile has a star. And I save the project file in that state. When I load it pack up, it should be in that same state

Yeah as far as I know that's precisely what happens. When you load the profile back up, it'll have the same base profile and again with a star because it'll also load some setting adjustments that you didn't save to the profile. As long as you don't modify the profile in your Cura, loading a project file will also not modify the profile. Though it will still ask you to override or clone, even if the profiles are exactly the same since this feature request is not implemented yet: https://github.com/Ultimaker/Cura/issues/6746

TKDesigns, 2019-12-11:
> Yeah as far as I know that's precisely what happens. When you load the profile back up, it'll have the same base profile and again with a star because it'll also load some setting adjustments that [...]

Correct. But it's also saving the custom profile in whatever state it's in at the time of saving the project. So any changes that are saved to the profile down the road are overwritten when loading an old project. Which it really shouldn't.

Ghostkeeper, 2019-12-11:
So you basically don't want to take the profiles from the project file then but just load the model data?

Or is the feature request to allow choosing to discard parts of the project file when opening it, instead of restoring the entire state?

TKDesigns, 2019-12-11:
> So you basically don't want to take the profiles from the project file then but just load the model data?
> 
> Or is the feature request to allow choosing to discard parts of the project file when opening it, instead of restoring the entire state?

What it should do is load a project with whatever settings you saved it with. If it matches you "current" saved custom profile, then it opens with no star. If it doesn't match, then it should open with all the settings and a star. And the "current" saved custom profile is still in the profile settings menu. Right now there is no way to load your custom profile on an old project if you had changed the custom profile since saving the old project file.

Ghostkeeper, 2019-12-12:
> If it doesn't match, then it should open with all the settings and a star.

So with this you mean that it sort of unsaves all of the most recent changes. I think that would be very confusing for the user especially if they also had other unsaved changes.

> Right now there is no way to load your custom profile on an old project if you had changed the custom profile since saving the old project file.

Yes there is. In the project opening menu, you select to create a new profile, then switch to the old profile after loading.

TKDesigns, 2019-12-12:
> So with this you mean that it sort of unsaves all of the most recent changes. I think that would be very confusing for the user especially if they also had other unsaved changes.
>
>
> Yes there is. In [...]

It doesn't unsave since I saved the project as a star anyways. So that doesn't really matter. I can keep trying to explain why this is wrong and a real pain when loading old projects. Creating new profiles for every old project means I'll have to sort through a ton of profiles every time I want to load one. It's backwards. Bottom line is a custom profile should NEVER change no matter if you start a new or load an old project unless I re save it...... It should be the most up to date version no matter what.

Ghostkeeper, 2019-12-13:
> It doesn't unsave since I saved the project as a star anyways. So that doesn't really matter.

What I think you mean is that it moves some of the settings from the profile in the project into the "unsaved changes"; those changes that have not yet been saved to a custom profile.

I think this is a matter of which workflow is more common and also which workflow is easier or more clear then. Because some people would expect it to restore the profile from the project and depend on that. Some people even use projects to share configurations with each other.

TKDesigns, 2019-12-13:
> What I think you mean is that it moves some of the settings from the profile in the project into the "unsaved changes"; those changes that have not yet been saved to a custom profile.
>
> I think this [...]

See, the problem is, especially when just starting out, you make a custom profile with certain settings. Then as you learn and or modify your 3d printer, this custom profile changes and updates. If you ever go back to look at an older project it loads this outdated profile and there is no way to look at the newer updated one when you're in the old project. So now your stuck guessing what settings you had updated. I don't see how the majority of everyone would want this. But it's obviously not going to change by the way I have to keep trying to explain this. Sooooo........ Thanks for your time and wasting mine. I guess I'll have to go through all my files and re-save them.

TKDesigns, 2019-12-20:
Oh another issue with this. When loading an old project, even after selecting "new project" or even closing cura and reopening, it's still loading an old version of the custom profiles...... It's really pathetic that you can't see how wrong this is.

Ghostkeeper, 2019-12-23:
> Oh another issue with this. When loading an old project, even after selecting "new project" or even closing cura and reopening, it's still loading an old version of the custom profiles......

Yes, if you've overwritten the profile with the version from a project (old or new doesn't matter; the project doesn't record at what time it was saved) then the profile is overwritten. If you wanted to have a back-up then you need to have it saved as a new profile instead. Then you'll have two profiles, an old one and a new one.

closed 2020-06-19 by no-response[bot]