Ultimaker/Cura #6739: Post-processing scripts are not removed when closing the program

ddbrown30, 2019-12-01:
**Application version**
(The version of the application this issue occurs with.)
Cura 4.4

**Platform**
Windows 10

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Add anything to the plate
2. Select Extensions -> Post Processing -> Modify G-Code
3. Select Add a script
4. Select any script (I'm using Filament Change)
5. Select Close
6. Exit Cura
7. Open Cura
8. Add something to the plate

**Actual results**
At this point, the script that was added will still be there

**Expected results**
All scripts should be removed when exiting the program.

**Additional information**
This is debatable as to whether this qualifies as a bug, but it has continually caused me issues. I'll print with filament change enable on one print, come back another day to do a different print, and filament change will still be enabled. Then I'll walk away from the print only to come back later to find it's paused in the middle waiting for the new filament.

nallath, 2019-12-02:
Ironically, the behavior that you describe as the expected result used to be the default. But we got a lot of people complaining about this, so we added a feature so that it does this.

ddbrown30, 2019-12-02:
The solution might be to make it part of the print profile then, rather than a floating, nebulous feature. It would make Cura more full featured by allowing you to set up multiple preset scripts that are associated with print settings without having to tear down and setup the scripts every time you need them.

Ghostkeeper, 2019-12-05:
> The solution might be to make it part of the print profile then, rather than a floating, nebulous feature.

We've decided to make it part of the added printer instead, since a very common use case is to adjust for printer incompatibilities. It's uncommon to use post-processing scripts together with certain layer heights or print speeds, as most of the scripts involve either print-specific tweaks (such as pause at height) or printer-specific tweaks (such as probe measurements or search and replace).

For your use case, you could simply add your printer twice to Cura, one with those post-processing scripts and one without.

We'll not change this, sorry.

closed 2019-12-05 by Ghostkeeper