Ultimaker/Cura #6740: se pone lento y se bloquea cuando se selecciona una pieza y se le da opcion de multiplicar

jhonraz66, 2019-12-01:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

**Reproduction steps**
1. Something you did.
2. Something you did next.

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

nallath, 2019-12-02:
Could you please fill in the bug template (and do so in English)?

Ghostkeeper, 2019-12-05:
Closing this for lack of information.

closed 2019-12-05 by Ghostkeeper