Ultimaker/Cura #6742: Object does not start on platform (no raft) when printed with another object

johnmhancock, 2019-12-01:
**Application version**
4.4.0

**Platform**
macOS 10.14.6

**Printer**
Ultimaker 3

**Reproduction steps**
1. Open print.ply in printer and sliced.  Was able to generate raft
2. Added print2.ply to print job and sliced.   Only raft was generated for print2.ply.

**Actual results**
1. Raft generated for print.ply
2. Raft generated for print2.ply but not print.ply

**Expected results**
Rafts generated for both objects.

**Additional information**
Attached are the original files and save of the print job.
[printjobmissingraft.zip](https://github.com/Ultimaker/Cura/files/3908676/printjobmissingraft.zip)

smartavionics, 2019-12-02:
The 2nd object was slightly above the build plate. I lowered it and now it puts rafts under both...

[image]

johnmhancock, 2019-12-02:
I'm not able to replicate this using the lay flat option in cura.  I am able to get a raft by changing object size to under 4000% but I would like to know if it can be made to work without manual tweaks, especially after it does generate the raft initially without any user intervention.  Should I expect an object placed in cura to be on the build plate when I add it, and then stay on the build plate when another object is added?

Ghostkeeper, 2019-12-05:
It's laying it flat properly, but the bottom side of the model is not flat. There's a bit under the leftmost dimple where model was not cleaned properly after the scan. You can see this when you reduce the line width and/or layer height to something extremely small in Cura (like 0.001mm). The inaccuracy is too small to print, but lay flat still recognises it.

[image]
[image]

So it looks like this is expected behaviour. You can fix the problem by moving the object slightly into the build plate. The mesh will clip the build plate then and will not be printed completely then, but in this case you probably don't want to print that inaccuracy anyway.

closed 2019-12-05 by Ghostkeeper

johnmhancock, 2019-12-06:
The object doesn't care which side is aligned with the build plate - the issue occurs whether the irregularity from the model is facing the plate or not.  It also generates the raft if it is the only object on the surface - should be behaviour be different depending on the presence of another object?  Why does the irregularity stop the raft generation if it is not on the surface used on the build plate?

johnmhancock, 2019-12-06:
I think I see it now - **all** sides have irregularities that interfere with raft creation, which cause the software to fail to start the object on the plate with a raft when a second object is present.  If I scale the object down, the irregularities become too small to affect the print and it makes the raft (under 4000%).  If I scale the object up the software is unable to create a raft but instead starts printing on the next layer.  If I generate support both models have a raft created for them.

I'm not sure what the correct fix for this issue is - it would be nice to have a mechanism to inform users their prints will fail without support and that Cura is going to start printing above the platform for an object - that the part piece at plate level is too small to print, and will be ignored.

Ghostkeeper, 2019-12-09:
One way to maybe notify users better is to NOT colour the surfaces red that are overhanging less than half the initial layer height above the surface. Currently the bottom side of your print is basically always red, which defeats the purpose of the warning colour there. If we were to make that more reliable, perhaps you could see more easily that it's going to fail to print there.