Ultimaker/Cura #6745: [4.4.0] Please add Creality CR-10 V2 profile

eXistenZ, 2019-12-02:
Hey, please add a profile for Creality CR-10 V2 printer! Or I can just use the old Creality CR-10 profile and it will work fine?

Thanks!

Ellecross, 2019-12-03:
That would be great! But we as Ultimaker do not create them. If someone creates a pull request, then we'll add it. ;)

Ghostkeeper, 2019-12-05:
Can't fix this since we don't have your printer model in the office. It's impossible to buy all printer models out there. If someone wants to make profiles for your printer, we'll gladly add them to Cura. But we'll close this bug report until then because there is nothing we can really do about it.

closed 2019-12-05 by Ghostkeeper

kubilesti, 2020-05-06:
what if someone sent you the printer and you returned it afterwards?

Ghostkeeper, 2020-05-11:
Then it's a matter of us finding the heart and time to create profiles for you in their free time.