Ultimaker/Cura #6746: Provide option to not overwrite, but still be based on, existing profile when loading project

smartavionics, 2019-12-02:
**Is your feature request related to a problem? Please describe.**

At the moment (master, 4.4.0), if you save a project that's based on some profile, (e.g. profile Foo) and then you modify Foo in some way and save that profile, when you next load the project that's based on Foo it gives you the option of either overwriting Foo (which loses the change that was made to Foo after the project was saved), or creating a new profile which is no longer based on Foo (I think) so the change to Foo is not visible in the loaded project.

**Describe the solution you'd like**

What would be good is that when the project that was based on the original Foo profile is loaded, it gives you a further option of using the new Foo profile (unaltered) and with the differences between Foo and the project's settings being visible just as if you had edited the settings (i.e. the profile name box has a * and the settings that differ from the profile are marked as we would expect.) The point here is that a difference in the settings between the project and the profile can come from either the user editing the settings or the profile being edited and saved before the project is reloaded.

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
Everyone.

**Additional context**

Wasn't this how it worked before? I can't remember!

Ghostkeeper, 2019-12-05:
The settings that are saved to your profile and the settings which are not are kept separate and I think we all want to keep it that way. The * near the profile drop-down just means that there are setting adjustments that are not saved to the profile, so that part should probably stay the way it is.

The rest of the feature request is still valid though. It never worked that way before.

The details of this feature still need to be thought out though.
A requirement of this feature is that setting modifications are saved with a date, so that Cura can know which setting modifications are made after saving the project file and which are saved before. To illustrate, imagine the following sequence of events:

1. On Monday, Suzy sets the Print Speed to 10mm/s and saves this as a profile. She shares this profile with Robin.
2. On Tuesday, Robin makes an adjustment to the profile to set the Print Speed to 20mm/s and saves it into the profile.
3. On Wednesday, Suzy makes an adjustment to the profile to set the Print Speed to 30mm/s and saves it into the profile.
4. On Thursday, Suzy gets the profile from Robin and imports it into Cura.

What happens now? Does it take the adjustment that Robin made, or does it override the value from the profile with the adjustment that Suzy made? Does it matter that Suzy made the changes later even though they were on completely different computers?

Cura has no notion of quality changes profiles being based on other quality changes profiles. When you select that they are not overriding, it copies the profile to a different name.

smartavionics, 2019-12-05:
I think I would be happy with the following behaviour... 

When you load a project file it would see if the project is based on a profile that already exists. If it isn't, it simply creates a profile with the name used in the project and stores the values currently used in the project into that profile. But if the project is based on an existing profile it compares the values used in the project with the values in the existing profile and if any differ it displays the * and indicates which settings differ from the existing profile. The user then has the same options as they do  now to either overwrite the profile using the values in the project or to discard the differences. I don't see why profiles need to have a date/time stamp.

Ghostkeeper, 2019-12-06:
We're mixing up some terminology here. A project is a file that contains multiple profiles (as well as some 3D mesh data optionally). One of these profiles is the "user changes profile" or as I prefer to call it the "custom profile". So going through your desired behaviour:

> When you load a project file it would see if the project is based on a profile that already exists. If it isn't, it simply creates a profile with the name used in the project and stores the values currently used in the project into that profile.

This should already happen. If there is no profile with the same ID, it will still show the dialogue but there will be no drop-down to change behaviour. It will always create a new custom profile with the values stored in the custom profile from the project.

> But if the project is based on an existing profile

Projects are always based on existing profiles, so I think you mean here that "if the project contains a custom profile with an ID that already exists in Cura".

> But if the project is based on an existing profile it compares the values used in the project with the values in the existing profile and if any differ it displays the * and indicates which settings differ from the existing profile.

Yeah currently it doesn't display which settings are different (that would make the dialogue appear like a spreadsheet and overloaded with information) but it only indicates that there is a collision.

> The user then has the same options as they do now to either overwrite the profile using the values in the project or to discard the differences.

Right, so what you're asking is not any new functionality with overwriting, but still be based on an existing profile. What you're asking is *only to display more information in the project open screen*. It should say that the colliding profiles are different and which settings are different.

smartavionics, 2019-12-06:
Slight confusion, I think. I am not asking for the settings that are different when a project is loaded to be displayed any differently to how they are now. I'm not asking for the project load dialog to show those differences. I simply want Cura to not overwrite an existing profile when a project is loaded that is based on that profile.

I would like Cura to compare the existing profile to the settings that are in the project being loaded (which, I assume, are the combination of the profile that was saved in the project file along with any unsaved changes) and if there are any differences, the * is displayed and those settings that are different are shown in the normal way in the settings pane.

I appreciate that a project is always based on an existing profile but when Bob sends a project file to Alice, she may or may not already have that profile. If she doesn't, Cura can just create it. If she does, Cura should not overwrite her profile as it does now, it should do what I have outlined above.

Ghostkeeper, 2019-12-09:
Ah right. Currently Cura checks if there are any colliding profiles (same ID) and when there is one, it'll ask whether to overwrite that profile or create a new one. You're saying that when there is a collision but the profiles happen to be the same (same setting overrides, same values) then Cura should not even ask?

The settings that the user had active when the project was saved are stored separately and restored upon loading it. So if you had a custom profile that changed the infill density and also a changed setting for Enable Support, and then saved a project, then the other person will also get a custom profile that changes the infill density as well as an unsaved setting that enables support (and a star will be displayed).

smartavionics, 2019-12-09:
> You're saying that when there is a collision but the profiles happen to be the same (same setting overrides, same values) then Cura should not even ask?

That's right it doesn't need to ask if the profile in the project being loaded matches the existing profile with the same id.

> So if you had a custom profile that changed the infill density and also a changed setting for Enable Support, and then saved a project, then the other person will also get a custom profile that [...]

Yes, and their custom profile is not automatically overwritten until they decide that they wish to save any unsaved settings that occur when loading the project. And just to make absolutely clear, an unsaved setting can occur when either the .3mf file was written with unsaved settings OR when the .3mf file is loaded into a Cura that already has a profile with the same id and any of the settings in the original profile collide with the profile + unsaved settings that arrived in the .3mf file.

Ellecross, 2019-12-10:
For reference, see CURA-7036

alsmithson, 2019-12-14:
Cura’s profile handling is confusing and I have several times overwritten stuff I wanted to keep. Then I started allowing projects to create new profiles every time it loads and I don’t know if they are the same or different. And unlike IdeaMaker, Cura has no way to compare profiles.

These two quotes from @Ghostkeeper would be a great step in the right direction and I would like to add a +1:

1. “It should say that the colliding profiles are different and which settings are different.”

2. “…When there is a collision but the profiles happen to be the same (same setting overrides, same values) then Cura should not even ask?”

yet-another-average-joe, 2019-12-20:
(hoping I'm not creating a "XY problem" !)

Maybe I have a workaround : my (tedious) workflow...

The profile management is sometime a bit confusing. So, everytime I work on a new project, I create a new profile for this new project. I then export the .curaprofile, give it the project name and save it along with the 3MF file. When the project is completed, I remove this profile from Cura (otherwisee the profile list would be a mess). If I reopen the project later, I also import the profile form the settings dialog box. The profile then appears with the project name. OK, it's a bit redundant, and really tedious ! (.curaprofile file is saved for display in combo box)

This way it's (more or less) easy to compare the most recent profile with the old project profile, and I can switch easily between old and new profile (this rarely happens as most of my projects need different profiles)

Redundant, somewhat tedious, but _in fine_ handy.

Maybe Cura could offer the option to save/load a .curaprofile with the project (.3MF) name, and/or Cura could create a temp profile when loading a project, and that temp profile would be displayed (as the project name) in the profiles combo box. The project profile being added to the profiles list, it would allow the user to switch/compare. This way, there would be no need to ask for keep/discard. The current profile would remain untouched, the old one would be temporarily added to the list and selected by default.

Ghostkeeper, 2019-12-20:
The project file contains the .curaprofile in a way. Both the Cura project file and the .curaprofile are actually .zip files that you can open with a zip archiver and everything in the .curaprofile is also in the project file. Loading a project file asks you if you want to save that .curaprofile to a separate profile, or overwrite the existing profile with the same name. So if you save it to a separate profile, that should be the same as when you load in a separate .curaprofile that was created at the time when the project file was saved.

yet-another-average-joe, 2019-12-20:
Yes, indeed, this is why I wrote that what I'm doing is redundant.

The goal is to have the project profile with the project name in the profile list, and at first let the current profile unchanged : it can be selected later if needed. This also allows to keep trace of multiple profiles for a single common project when testing parameters, without keeping a pile of configurations in the system directory/combo box for the rest of the life. It can also be done at Save time by just saving the 3mf with different names, but in this case, changing "builplate parameters" will do it for the current 3mf file only... I find playing with .curaprofile files on only one buildplate is very usefull.

A pratical example : a model with orientation, rotation, scaling, custom supports, etc, and testing multiple profiles (material, shell, infill, etc.) based on a standard one : only one profile can be saved with the project file. Some CAD softwares can store different configurations for the same model. External .curaprofile files mimics this functionnality, and the "Manage profiles" in the profiles combo box is there ! Probably not the way it was intended to be used, but it makes it possible.

A few months ago, I was about to issue a FQ for Cura to fill the exported profile edit field with the project name, pointing to the project directory (the field is empty and it exports to the last export directory by default). Not an easy subject of discussion, as everyone seems to have a different workflow ! Making things easier for some users could make things more difficult for some others.

iRoc999, 2020-11-22:
For simplicity how about when a project file is opened Cura automatically creates a temp profile and never touches saved profiles EVER. That way opening an older project doesn't destroy months of tuning, because that's exactly what it does in current form due to the seriously [redacted] way the dialogue is worded. 

it's simple open project/ new profile is created with project name. If not saved upon close it vanishes and leaves existing profiles untouched. This way if the user has made adjustments to their profile they can simply select that profile. If someone else is opening the project they can just select one of the profiles that they know work well on their machine. No confusing dialogue, just an opened project ready to be sliced with what ever profile the users selects, but also includes the temp profile with project name for instances where someone is remotely helping someone else tune settings. 

Also could just change the wording to
Use included profile ( create Temp Profile )
choose existing profile from profile manager  * and then the profile manager pops up so user can SAFELY choose the profile they want/need

Ghostkeeper, 2020-11-26:
Cura makes the distinction between custom profiles and setting changes that have not yet been saved to a profile.

For custom profiles the behaviour is already what you are asking: It's going to create a new profile for it. To prevent creating hundreds of profiles by re-opening the same project file it'll ask to adjust the existing profile if the profile name is the same, or to create a new profile for it. The difference between the current behaviour and what you're asking is really that the new profile is not temporary. It couldn't be, because you can select that profile in any of your added printers and making it disappear would require those printers to change to a different profile.

Unsaved setting changes will be overridden if you select to re-use an existing printer. They will be retained if you select to create a new printer. Again, similarly, you don't want it to keep creating new printers for every project file that you load.

The dialogue is confusing, in part because everyone expects different wordings there. We've adjusted the wording in 4.7 but apparently it still produces expletives from people who expect it to say "temp profile" for profiles that are not temporary. You can SAFELY choose the profile you want or need right now. The problem is user experience.

iRoc999, 2020-11-27:
I can feel your pain as through researching the issue i can see it's been an issue for quite awhile. With that said I feel there's a communication issue. It really seems that users are continually asking for a 3rd option of "open project create no new profile and make no changes to existing profile" that just imports the project models on the build plate. We just haven't communicated that well, likely due to the fact that when we encounter the issue it causes damage to tuning progress or causes us to have to do profile clean up. Which in turn causes frustration and that leads to not thinking clearly

Majority of us understand the utility of printer settings being able to be saved, and shared with the project, but printers change. By that I mean printers get altered through necessity or whim, and the profiles also evolve.
Currently one can only update the existing profile wrecking any tuning that's been done or create a new profile and then have to clean up a growing list of unwanted profiles

I guess in summation a 3rd option of "just import models" really seems to be the magical solution
My gut is telling me that if that were included and a tick box for remember these choices and don't show this dialogue again were added you might find that gets done more often that not. That is if there's usage data collected....
This option of "don't show this dialogue again" would need to be able to be turned back on in preferences though as there is a legitimate need for profile importation

nallath, 2020-11-27:
> It really seems that users are continually asking for a 3rd option of "open project create no new profile and make no changes to existing profile" that just imports the project models on the build plate. 

But that option already exists: 
[image]

Felix-ArduPilot, 2020-12-20:
As far as I understand that, the “Import models” option does only import the .stl without profile, correct? I am often running into the problem, that I want to print an old model with the same settings I used when I printed it the first time. So I need Cura to open the profile, that is included in the .3mf file. But I don’t want it to override or change my current profiles, because I would lose all the tuning I have done in the meantime. And I also don’t want it to create a new profile, because after opening a number of files, that would get really messy and I wouldn’t know which one are my profiles, which one are the profiles, Cura created, which one is old, which one new…

But I also was thinking about creating a temporary profile, like @iRoc999 already suggested. I think it would be the best to just call it “temp”. So everyone would know that it is the profile that came with the .3mf file. If you like to print the model with the old settings, just hit print. If you want to use your new profile, you can select that (and be sure, that it was not changed by loading other .3mf files). And after you close the file, the “temp” profile can be deleted and all the settings are the same as when you opened the project file.

Ghostkeeper, 2020-12-23:
> As far as I understand that, the “Import models” option does only import the .stl without profile, correct?

Aside from technicalities of it not being .stl but .3mf, yes. It just imports the models as if you're loading any other file format like STL.

Cura doesn't know when you're done with a temporary profile. You want to print a model with it, okay, but if you then make modifications to the settings and print again, it needs to not delete the printer in front of your face saying you're only allowed one print with those settings of course. And if you close Cura and expect the configuration to be deleted upon restart, any modifications you made would also be lost, which may cost time you've diligently spent on updating profiles.
So instead of guessing when the user is done with the profile, we added a convenient button by which the user may indicate that they're done with the project. It's the "Delete" button in the printer management page.

Felix-ArduPilot, 2020-12-23:
> Aside from technicalities of it not being .stl but .3mf, yes.

Okay, thanks for the clarification.
I don’t really get the problem with deleting the temporary profile. I think it would be the best to create a “temp” profile with the settings stored in the .3mf file. Then you can print it with those settings. Or you can make modifications. If you want to keep them, you can either save them by overwriting the .3mf file of that specific model or you can give the temporary profile a name of your choice, so it is saved. When you close Cura the “temp”-profile is deleted (or if you like it better just overwritten when you open the next .3mf file)

My problem with the current situation is, that I am always really confused about what is stored in a profile. I have basically one profile which evolves more and more over time. Now when I open an old .3mf file months of tuning may be destroyed. I think when a user has manually saved a profile, he wants it to stay like that forever and Cura shouldn’t change things in that profile. Otherwise the user will be completely confused. At least that’s how I feel about it.

I am aware of the fact that it is a really difficult task to find a solution that fits the needs of all users. So other people might see that differently. But that would be my suggestion for it.

Ghostkeeper, 2020-12-28:
I understand that you're confused about what's in a "profile". The terminology is complex too. It seems you've switched in your last comment the meaning from a printer instance to a custom profile. Cura has many types of profiles: Printer definitions, machine-changes profiles, nozzle profiles, material profiles, custom profiles, quality profiles, intent profiles, printer instances and user-changes profiles. We keep this terminology internal since no novice should need to understand it to be able to use Cura, but it makes it quite difficult to interpret your feature request and even more difficult to communicate our interpretation of it or any problems we foresee.

Loading a project file will cause it to load a printer instance (an "added printer") and set all of the profiles on that printer instance to what they were when the project was saved. So it currently either adds a new printer instance or allows you to override one of the existing ones. And then on that printer instance it adds the correct profiles. Built-in profiles (definitions, nozzles, materials, quality and intent) are always re-used. Material and quality changes it asks the user whether they want to override the existing one with the same name or create a new one. Machine-changes and user-changes profiles are tied one-on-one to the printer instance; it overrides the existing values if re-using a printer instance.

So to have "temporary" profiles we'd need to choose which of these are temporary and what happens when you delete them. Since user-changes and machine-changes are tied to the printer instance, making the instance temporary would delete any changes you make to the printer upon restarting Cura. Especially since Cura currently saves the full state, users wouldn't initially understand why their hours of work are lost. And in the long term, it'd be quite hard to make them understand which of their printer instances are going to disappear at the restart of Cura and which are going to stay. It's generally complex to understand that some printers may disappear and this is a dangerous game when working with users' hard work. I'd be very surprised if we didn't see a number of bug reports about users' printers disappearing if we implement that.

The problem exists with other types of profiles as well, like a custom profile or a user-created material profile. Loading loads of 3MF files and creating a new custom profile for them mutters up the list of profiles there too. This comes with the additional challenge that these temporary profiles can be used in non-temporary contexts. You could activate the temporary custom profile in a different printer which is not temporary. Deleting that temporary custom profile then would corrupt the other printer then.

Felix-ArduPilot, 2020-12-29:
Thanks, I really appreciate your help! Yes, you are right, terminology is really complex here. I am just a normal end-user and definitely no Cura-expert, so please forgive me my mistakes. But I really like Cura and I use it for some years now, so I thought, maybe it can help if I share my experiences and the things I’m still struggling with.

In my last post, I actually meant both, printer profiles and custom profiles since Cura handles them in a relatively similar way, that can create confusion (at least for me). I think the most important improvement would be that a user stored profile (no matter if it is a printer profile or custom profile or whatever) can never be changed by Cura itself.

An example: I have tuned my printer settings (things like print speed, bed temperature and so on) really well now and I save it as a custom profile called “Felix latest”. Now I open an old .3mf file for printing the object exactly as I printed it the first time. Of course some settings may have changed since then, but I want to use the old settings because e.g. I made some model-specific settings (like support adjustments or things like that) and I know it has printed well back then. Now Cura lets me choose: It can create a new profile (which I definitely don’t want, because I would end up having dozens of profiles named in a very similar way) or I can update the existing profile. But actually updating is a nice word for overwriting. So if I design a new model and want to print it with the ”Felix latest” profile that I just saved, it is actually a very old profile coming from the old .3mf file. So months of tuning would be lost.

So my feature request would be: If a user has stored a profile (printer profile, custom profile etc.) never let Cura change or overwrite it. It should be permanently stored.

Now how could we achieve that? I understand the problem with deleting temporary profiles. We would need a very clean separation between what is temporary and what is permanently stored. So one way would be to create a temporary printer instance with all other related profiles (custom profile etc.) also being temporary. But the user would need to understand, that all changes he is making in these profiles are deleted when he closes Cura. (Of course he or she can still save them by renaming and saving the profile manually.)

Another way I can think of would be the following: Cura is creating such a temporary printer instance (with all other types of profiles) once. But it is never deleted. Instead it is overwritten (“updated”) when a .3mf file is opened. Basically the same way as it is right now. With the difference that only the temp. profile is overwritten and never a user saved profile. That way a user who doesn’t understand the concept wouldn’t lose his settings, but also all manually saved profiles stay the same, no matter what .3mf file was opened before. What do you think about it?

Ghostkeeper, 2021-01-06:
> But the user would need to understand, that all changes he is making in these profiles are deleted when he closes Cura. (Of course he or she can still save them by renaming and saving the profile manually.)

I think this is where the difficulty will be. If the user doesn't know that the profile will be deleted, he'll be confused as to why the settings they worked on yesterday are gone. It's a similar problem to what you're experiencing.
Cura currently stores everything when you close the application, except the contents of the build plate, so for current users this would not be expected. Users tend to not read or remember text that's more than a few words, so a pop-up would only be annoying. The profile name is quite easily missed. I don't really see a way to make this intuitively make sense. And aside from that, the profile can still be loaded by other printers that are not deleted and deleting a profile that's in use currently corrupts the printer (we disable the Delete button).

The alternative you proposed doesn't have those problems. Currently you can do this in Cura by manually selecting the same profile to override every time you load the project file. It does have the problem that we'd need to somehow make clear that those profiles will be overridden by project files always so they'd need to duplicate the profile in order to keep it. It's really just the same as how Cura behaves now but the drop-down would always default to the same profile and you don't get control any more over which profile is overridden.

Both solutions have the problem that people also use project files just to share a set of matching profiles with each other. This would then require more steps (somehow making the project file's profiles permanent) and this is not intuitive to them then. It kills one of the use cases for project files.

Felix-ArduPilot, 2021-01-10:
Thanks a lot for looking so deeply into my question. I really appreciate your help! And it feels like we’re getting closer and closer to a solution now.
> If the user doesn't know that the profile will be deleted, he'll be confused as to why the settings they worked on yesterday are gone.

I totally agree, that nothing should be added that might confuse a user. The main objective of this discussion is to reduce all points of confusion as much as possible. So let’s stick with way number two which was updating the “temp. profile” instead of deleting it.

> The alternative you proposed doesn't have those problems. Currently you can do this in Cura by manually selecting the same profile to override every time you load the project file.

That is true, but the dropdown menu doesn’t let me choose the profile which means, that I would have to open Cura, load the profile that I would like to overwrite, close it again and then open the .3mf file and chose “update existing” in the drop down menu. That’s a bit complicated. But yes, it is possible right now.

> It's really just the same as how Cura behaves now but the drop-down would always default to the same profile and you don't get control any more over which profile is overridden.

Yes, but that would be essential. If a user saves a profile, he wants it to stay unchanged. Cura is interfering with that wish right now and doing things on its own. Things that most of the standard users can’t fully understand. At least I can’t. Open like five .3mf files of a year ago and tell me what the profile settings are after that. That would be at least really difficult. I think it also depends on which instance of Cura you close last and things like that. I think the only way to solve that problem would be to stop Cura from modifying user-saved profiles at all.

>  It does have the problem that we'd need to somehow make clear that those profiles will be overridden by project files always so they'd need to duplicate the profile in order to keep it.

What if the entry dialog was left out completely? Instead the temporary profile could be updated and when the user closes Cura he will be asked if he wants to save it. Just like in a word document or with most other filetypes. He or she can then overwrite an existing profile or create a new one or click on “don’t save” if he just wants to exit.

I often experience the following scenario: I slice a new model and remember that I once printed a model with relatively similar problems already. So I want to know things like “what was the initial layer speed I used for the old model?” I just want to look this one parameter up in my .3mf file, but I don’t want it to change anything in my Cura profiles. That’s very difficult right now and I think most other users feel the same. So it would be a huge improvement, if we could find a solution.

> Both solutions have the problem that people also use project files just to share a set of matching profiles with each other.

I think that problem would also be solved by using an exit dialog instead of an entry dialog. That wouldn’t require more steps for the user, but it would be a clear statement. “Save” “Don’t save”. And when “save” is clicked you can chose a name or overwrite an existing profile. And also no printer can be corrupted, because the temporary profile in the background is never deleted, just updated.

Ghostkeeper, 2021-01-18:
> And it feels like we’re getting closer and closer to a solution now.

That's the best part of these discussion-ish threads. You're kind of working together towards a solution. Free software is not just open source code but also about collaboration in other types of development.

I think I've been misremembering things a bit. In [this recent pull request](https://github.com/Ultimaker/Cura/pull/8144) we've made it possible to select which printer to override. I thought this applied to each of the drop-downs, but apparently it applies only to the printer drop-down. For the other ones it's still just a choice between "update existing" and "create new". It would be good to make that consistent and allow you to choose which profile to override there too.

> What if the entry dialog was left out completely? Instead the temporary profile could be updated and when the user closes Cura he will be asked if he wants to save it.
> [...]
> I think that problem would also be solved by using an exit dialog instead of an entry dialog. That wouldn’t require more steps for the user, but it would be a clear statement. “Save” “Don’t save”. And when “save” is clicked you can chose a name or overwrite an existing profile. And also no printer can be corrupted, because the temporary profile in the background is never deleted, just updated.

Our UX engineer is looking into stuff like this at the moment, actually. He doesn't have a Github account so I can't ping him here, but he should be looking at UX-marked threads. I'll point him to it again.

Felix-ArduPilot, 2021-01-19:
> That's the best part of these discussion-ish threads. You're kind of working together towards a solution. Free software is not just open source code but also about collaboration in other types of development.

That is true and it’s great to see how well this concept works. There are a lot of very smart people all over the world contributing to the project and working together. Without that, awesome programs like Cura wouldn’t be possible! So thank you all for your effort!!

> In [this recent pull request](https://github.com/Ultimaker/Cura/pull/8144) we've made it possible to select which printer to override. I thought this applied to each of the drop-downs, but [...]

Ah, thanks for mentioning that. Yes, having the possibility to choose which profile to override would definitely be an improvement.
> Our UX engineer is looking into stuff like this at the moment, actually.

Thank you very much. I really appreciate your and his help! :)

Triangulix, 2022-06-13:
@smartavionics I feel your pain, too. The behavior is idiotic. I had several tickets around the unwanted overwriting of my profiles and printer settings when opening projects to print a part again and the developers always play the same script:

First they act as if we were talking nonsense and they had no idea what we are talking about.
Later they act as if the opening of the project wouldn't really overwrite the saved profiles on the harddisk.
After lots and lots of wasted clarification attempts the answer always is: "Yes, but we like it as it is."

We should accept that Cura cannot safely save and restore print projects. Period.

pengoses, 2022-11-09:
I wanted to add that I agree with iRoc999
When opening Cura project files, I **do** want to **temporarily** import the slicer settings, but I **don't** want any current slicer or printer profiles **ever** overwritten.

For those who do want that to happen, which I think would be rare, it should require some extra warning, an added checkbox near the OK button that warns "Warning: check this box to permanently overwrite your current settings!" If that's **not** checked, all machine/slicer settings loaded are temporary; not saved when Cura closes.

(I recently loaded a couple years old cura project file to reprint a part and ended up having my all tuning start gcode overwritten, which sounds like it happens to a lot of people with the current implementation.)

> For simplicity how about when a project file is opened Cura automatically creates a temp profile and never touches saved profiles EVER. That way opening an older project doesn't destroy months of tuning, because that's exactly what it does in current form due to the seriously [redacted] way the dialogue is worded.
> 
> Also could just change the wording to Use included profile ( create Temp Profile ) choose existing profile from profile manager * and then the profile manager pops up so user can SAFELY choose the profile they want/need

Triangulix, 2022-11-09:
> I wanted to add that I agree with @iRoc999 When opening Cura project files, I **do** want to **temporarily** import the slicer settings, but I **don't** want any current slicer or printer profiles [...]

I opened and commented on several tickets regarding this behavior over the last few years and as you see they close my tickets and delete my frustrated comments.

I have accepted that Cura cannot save and load projects in a non-dangerous manner. I have completely stopped to save projects. Intead I either save an additional profile when I expect to do the print again or I open the old GCODE file in a text editor and carry over the settings by hand.

Felix-ArduPilot, 2022-11-09:
> When opening Cura project files, I **do** want to **temporarily** import the slicer settings, but I **don't** want any current slicer or printer profiles **ever** overwritten.

I totally agree! Cura should NEVER overwrite a user-saved profile of any kind! That is the main cause of all this confusion.

Actually, I think the solution is pretty simple: Cura should open .3mf files as temporary profiles and when it is closed, the user should be asked to save the file: “Permanently save changes?” “Yes” “No” “Abort”.

That is a very simple handling mechanism, that everyone is already used to, because most other programs work like this. It’s exactly like writing in a text file: If you just write in the file, no one would expect the changes to be saved. And no one would expect fonts to be changed just by opening another file with other settings. So you have to make your changes and then hit the save button. And then this file stays like that until you change it yourself. The program will never change or overwrite things. And that is exactly how Cura should work in my opinion.