Ultimaker/Cura #6749: Cura version 4.4 Bad Gcode burying extruder into print

3dprinterhobby, 2019-12-03:
Cura 4.4

Have been using Cura older versions for past 1 1/2 and no issues.  this recent update has been plaguing me with issues.  Part way through the project i will either get and extra space + in z height or in some cases the code tells the nozzle to go down - in z height crashing or melting into the project

**Platform**
MAC OS High Sierra
version 10.13.6

**Printer**
Creality CR10s 

(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

**Reproduction steps**
1. save to SD card
2. Go to printer and start project.

**Actual results**
Part way through the project i will either get and extra space + in z height or in some cases the code tells the nozzle to go down - in z height crashing or melting into the project  have to force stop the project

**Expected results**
Expect to print as normally would

**Additional information**
Was unable to upload file here (55mb file), please provide instruction on how to upload file to you

Ghostkeeper, 2019-12-05:
If it happens with any object, you can upload a project file by loading in a simple 1cm cube or something and saving that project. That should only be a few kB then. If it only happens with that specific 55MB file you'll need to find a file sharing site like WeTransfer to share a project file.

Timocop, 2019-12-09:
I had this issue with my 'Geeetech i3 Prusa Pro W' 2 years ago in Cura 3.6 too. No post process used, minor config changes. It happend while z-hopping with retraction. After it crashed into the print and stopped the print, ive resliced the same model again with the same settings and it worked again, no suicide attempt. The GCode file that was on the SD Card was not corrupt.

Also im not sure if its related and i havnt checked the GCode yet but Cura 3.6 and higher seems to corrupt 1 layer sometimes when resilicing (e.g. drag&drop a STL onto Cura) while silicing already is in progress. Cura 4.2:
(Random middle layer placed on buildplate)
[image]

3dprinterhobby, 2019-12-09:
[Body.curaproject.3mf.stl.zip](https://github.com/Ultimaker/Cura/files/3937432/Body.curaproject.3mf.stl.zip)

3dprinterhobby, 2019-12-09:
@Timocop I had z-hop and retraction on too.  But so dar switching back to Cura 4.0.0 i have not had any issues.  Finished the print that had the error and just finished another print 21 hours later...no issues.

Timocop, 2019-12-09:
@3dprinterhobby It happend to me only once in those 2 years ive been printing. But the bug in the picture ive posted still happens sometimes. So i always reslice if that happens. I also turned off automatic slicing after this bug. When did your collision happen? On what percentage? Mine was like around ~50%.

Ghostkeeper, 2019-12-09:
@Timocop what you're seeing there in Layer View is a known bug just in layer view. The g-code is unaffected and the print head will not actually bury down into the print.
We've not been able to track down this bug. It happens for me maybe once a month (and I slice dozens of times per day for work) so when it happens I won't have a debug trace set up around layer view to see where things got corrupt. It seems as if the Z coordinate of the layer view data just gets set to 0 sometimes.

But anyway reslicing removes it and even if you don't reslice it still prints normally.

Timocop, 2019-12-09:
@Ghostkeeper Kinda spooked me that visual glitch i thought something bad is gonna happen if i print it like that. I guess its not related then. Thanks for the heads up 😅

Ghostkeeper, 2019-12-09:
> [Body.curaproject.3mf.stl.zip](https://github.com/Ultimaker/Cura/files/3937432/Body.curaproject.3mf.stl.zip)
> 
> @Ghostkeeper
> here is the file

This file is not a Cura project, nor a 3MF, nor an STL file as the title would suggest. Instead it's a zipped g-code file :smile: When loading that g-code into Repetier Host, it indeed shows that layer 1283 is displayed on the build plate. However I find nothing wrong in the actual g-code of that layer so I don't know how it arrives at that?

I've filtered that g-code with Grep to only show the lines with ` Z` in it to make it more handleable: 
[only_z.zip](https://github.com/Ultimaker/Cura/files/3940731/only_z.zip)

However that file shows only monotone increasing of the Z coordinate.
Grepping for `G92` only shows resetting of the `E` dimension, not the `Z`.

I'm grasping at air here. It must be in that g-code file somewhere but I can't find why it's going wrong...

Timocop, 2019-12-09:
Well on my end it wasnt only the Z issue. There where more issues than that.
I remember increasing the retraction speed up to Marlin limits 25mm/s and that where issues begun. Could also be a faulty SD card or Memory Overflow or what ever i have no idea.

Heres my story:
I had my printer next to me when it happend. Sometimes idk why but on my LCD screen it printed the opcode "M230" in plain text like in 1/2 of the print done normaly where "Printing..." should be. That happend more often too. Before the Z collision happend i saw that my printer had somekind of retraction issue 2 layers before the crash. It retracted a little more than it should. Then finally on the outer wall it retracted like 10cm or something like the Marlins 'Change filament' option. It fully pushed out the filament. It z-hopped far more in +Z like 1cm up after that and then moved fully down in Z until the nozzle was inside the models plastic. I shut off my printer immediately.

Since then ive changed my micro SD Card (with adapter) into a normal sized SD card and turned down my retraction speed to 20mm/s. (Also changed my extruder heater because it was dying / not heating properly)

Offtopic PSA: Enable thermal runaway protection in the firmware. Your heater may die in mid-print while youre asleep, then your extruder gets stuck because its cold and your extruder stepper motor driver chip dies because the extruder motor got stuck for hours 👍

3dprinterhobby, 2019-12-09:
[image]
@Ghostkeeper 
In error, like an idiot I first changed the file name to Zip, rather than use the compress action.  Hence why you confusion.
Just to give you a sense this the error product.  approx 63 mm in height.
I had another file this happened to 2 times.
1st time z- Collision approx 50mm height, took the plate right off the bed - i thought something just went wrong and started over withe same file
2nd time, z- collision at the same height.
3rd time - replaced and loaded the file, no issue
I deducted it to me maybe pulling the card while saving.  But then this issue was repeating.
Wish I kept that Gcode.

Ghostkeeper, 2019-12-10:
After saving to an SD card, Cura will show a button to eject the SD card. This button should be safe; it'll finish writing before completing the ejection. Did you try that?

3dprinterhobby, 2019-12-10:
Yes, I usually wait for that message.  That is my routine, I either use that option to eject or use the eject button in Finder (MAC).  it was just a theory on the first and second failure when this all began.  But that is now not the theory anymore as the problem has persisted.

closed 2020-06-19 by no-response[bot]