Ultimaker/Cura #6750: retraction_extra_prime_amount

maukcc, 2019-12-03:
In V4.4 when retraction_extra_prime_amount value is set in machine definition file, it is not taken into account. When retraction_extra_prime_amount value is set in a quality file, we even get an error.

smartavionics, 2019-12-03:
Hi, I think you need to fill out the details mentioned in the template otherwise this issue will not get much attention.

Ghostkeeper, 2019-12-05:
The `retraction_extra_prime_amount` setting is currently already in HMS434's quality profiles:

https://github.com/Ultimaker/Cura/blob/f7709c11bc2a28fa5b45ae97dbff8c7a1e301fbf/resources/quality/hms434/pla/hms434_0.4_pla_normal.inst.cfg#L29

This is most likely why changing it in the machine definition file has no effect. It's being overridden by the quality profile for PLA.

This is also why adding it again to the quality profiles gives an error. It will complain about the same setting having two different values in the same quality profile.

Let me know if I've made the wrong assumptions here (like the assumption that you're using PLA). But it looks like this is expected behaviour.

closed 2019-12-05 by Ghostkeeper

maukcc, 2019-12-06:
Aahh, we still had PLA quality profiles in there. I thought we had taken them out ages ago.
Can someone merge our commits please #6694