Ultimaker/Cura #6752: Cura 4.4 crash before launch

Tygopit, 2019-12-03:
Win 7 Pro with existing installation of Cura 4.3 beta.
Cura 4.3 runs as usual.
No start screen, just info that "Program has stopped working"
Before closing of the above info both the process and the app are present in Task Manager.

Log file:
Fatal Python error: Segmentation fault

Current thread 0x000016ac (most recent call first):
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGLContext.py", line 29 in setContext
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGLContext.py", line 106 in detectBestOpenGLVersion
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 133 in initialize
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 326 in initialize
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Application.py", line 215 in run
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 304 in run
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 745 in run
  File "X:\4.4-exe\build\inst\bin\cura_app.py", line 180 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

Any idea?
Cheers,
Tygopit

Ghostkeeper, 2019-12-05:
Could you try updating your graphics drivers? Since the error message indicates that something is going wrong upon creating an OpenGL context, it could be caused by your graphics drivers.

If I recall correctly, we've seen something similar that was also resolved by updating or reinstalling graphics drivers.

titan04, 2019-12-05:
Also seeing this issue when 4.3 was running just before.   New windows 10 vm with same behavior.  Current windows 10 system shows this with and without graphics upgrade and with window 10 update as well

bortek, 2019-12-11:
I'm having the same issue here.

bortek, 2019-12-11:
Uninstalled cura 4.4.0 including removal of config setting. Delete .cura dir. Updated Win10 to the latest. Updated Intel graphics 4000drivers. Reinstalled Cura 4.4.0. Same problem. Pushed send crash report button so you guys have logs.

Backing to 4.3.0 for now.

softtouch2009, 2019-12-23:
Same issue here. Latest NVidia driver, Windows 10 64 bit. All old cura versions run fine.

davidw3591, 2020-01-26:
Same issue. But 4.3.0 also doesnt work anymore since today?

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

Tygopit, 2024-11-22:
Please, close it.

closed 2024-11-22 by GregValiant