Ultimaker/Cura #6753: [4.xx] UM3 Network Printing Plugin makes Cura unbearably slow.

lbibass, 2019-12-03:
**4.x.x**
(The version of the application this issue occurs with.)

**MacOS / Windows**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)

**Ultimaker 3 Extended**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

**Reproduction steps**
1. just installing / having the plugin active makes cura impossible to use. 

**Actual results**
Cura takes forever to change the UI, load settings, slice, etc. 

**Expected results**
Cura actually works well. 

**Additional information**
No additional information.

Ghostkeeper, 2019-12-05:
Yeah we've seen this as well. It seems that the Zeroconf library we're using got broken at some point or used in the wrong way. It's spinlooping upon checking whether DNS queries are correct, taking 100% CPU power and locking up the GIL often.

lbibass, 2019-12-10:
Any updates on this? 

Thanks!

nallath, 2019-12-11:
We're still actively looking into it, but we haven't found a solution just yet.

macaba, 2019-12-12:
I was trying to find a temporary workaround for this on Windows as it was extremely painful to use the UI:
Go to `C:\Program Files\Ultimaker Cura 4.4\plugins` and delete the `UM3NetworkPrinting` folder. (Or move it somewhere else)

Ghostkeeper, 2019-12-13:
We've made a patch here to resolve this issue: https://github.com/jstasiak/python-zeroconf/pull/205

lbibass, 2019-12-13:
@Ghostkeeper great! That looks like a pretty noticeable improvement. How should I go about integrating those changes into my current installation of cura? Should I re-compile it with your changes?

Ghostkeeper, 2019-12-16:
Our current plan is to use Nallath's fork in our Cura builds until that PR has been merged and released by @jstasiak. If there is a release there we can use the mainline `python-zeroconf` package again. If you are able to build Cura yourself you can do that too once we've adjusted `cura-build-environment` to use Nallath's version (in a day or two I think). Otherwise you'll have to wait for the Cura 4.5 beta. I think that's due somewhere in February.

jstasiak, 2019-12-16:
For what it's worth I'm likely to review and release the python-zeroconf changes today.

jstasiak, 2019-12-16:
Just released python-zeroconf 0.24.1 (the least significant of the patches didn't make it, see https://github.com/jstasiak/python-zeroconf/pull/205#issuecomment-566131333).

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

lbibass, 2024-11-22:
> I'm cleaning house. Is this still an issue in current Cura versions (5.8.0 and up)? Can this be closed?

I pretty much don’t touch Cura anymore, but I have not had any problems with this for years.

closed 2024-11-22 by lbibass