Ultimaker/Cura #6755: Bridging is not bridging well

FlailAway, 2019-12-03:
**Application version**
4.4.0

**Platform**
Mint 19.3 + Cinnamon

**Printer**
Anycubic delta

0.4 Nozzle, PLA
I have a 2mm wide wall  using 3-WallLine Count and an SD card slot (28mm x 5mm) half way up the wall.
I am getting a very poor bridging using untouched factory settings in Experimental. The bridge is sagging around 3mm.

With so many options, I have no idea where to start changing. Would it be possible to have a sub-set of the most influential changes to the process, rather than that wall of options?

[image]

smartavionics, 2019-12-03:
Please save the project file (File -> Save) and then either zip or rename to zip and attach to this issue. Thanks.

FlailAway, 2019-12-03:
What is in a .3mf file? 
I'd like to know what I am putting here for anyone and everyone to download on a whim. I looked with a text Editor but too many wobbly characters so I assume it is mostly binary.

smartavionics, 2019-12-03:
A cura project file (.3mf) contains the model, the profile and other settings (setting visibility, etc.) It's actually a zip file so if you rename it to .zip you can then unzip it and check yourself.

FlailAway, 2019-12-03:
OK, thanks, just tried slicing that same thing with Prusa 2.1.xx and their factory settings. {gasp!!}

[image]

I will get the 3mf file and attach it in a minute, it is on a non-connected PC.

FlailAway, 2019-12-03:
Here ya go cld-1.zip -- and no, I am not going to use Prusa for the full print it has to many quirks to be useful.
[lcd-1.zip](https://github.com/Ultimaker/Cura/files/3918323/lcd-1.zip)

smartavionics, 2019-12-03:
Thanks for the file. I printed the bridge on my delta printer. To save time, I reduced the model by sinking it into the buildplate. My normal wall speed is 50 mm/S so the bridge walls were printed at 25 mm/s. The critical settings for the walls is speed, flow and fan.

[image]

This is how it came out, pretty good...

[image]

FlailAway, 2019-12-03:
OK, thanks, I will try it tomorrow, but I am not sure that your sample is a true representation with the wall starting from the brim and going up. That's not what I am trying to get. The Prusa version is good enough for me, but Cura is way off, so not sure you are there yet. :)

The only Bridge settings I can find for Prusa were Speed 80, fan 100 and nothing specific for flow, so I assume 100%.

I have just tried IdeaMaker and it is doing even better than Prusa. I can live with Ideamaker's foibles more so than Prusa so will use it for the full print. I do not want to run a 5-hour Cura print and find saggy Bridges as those slots are about 40-minutes from the end of the print.

smartavionics, 2019-12-03:
> OK, thanks, I will try it tomorrow, but I am not sure that your sample is a true representation with the wall starting from the brim and going up. That's not what I am trying to get. The Prusa version is good enough for me, but Cura is way off, so not sure you are there yet. :)

The fact that I didn't print the whole model makes no difference to the bridge quality. If anything, the bridge would have been worse due to the fact that it's nearer the heated bed. Anyway, if you use appropriate settings, Cura's bridging is very good. I'm leaving this issue now as I have nothing else to suggest. Good luck!

Ghostkeeper, 2019-12-05:
Since the Anycubic FDM printers are clones of Prusa's printers, it somehow makes sense that PrusaSlicer's defaults work better for Anycubic than Cura's defaults. In the case of the experimental bridging settings, they were tuned by Smartavionics since he made the feature, and optimised for his printer. The bridging settings in particular require a lot of tuning. This is the point of the bridging settings though: to allow that tuning.

I think to take the bridging settings out of experimental we'll need to make them a bit less overwhelming. Perhaps this can be done by grouping some settings together, sacrificing tweakability for simplicity. Perhaps we need to somehow gate the settings better for an easier learning curve. Definitely needs some UX love and thought from a process engineer.

Anyhow, I think that the original problem of FlailAway is a problem of creating correct profiles for his printer, not really a problem of Cura generating wrong toolpaths. So we'll close this. If you ever find better settings for your printer, please share them with us so that they can be added to a new Cura release!

closed 2019-12-05 by Ghostkeeper