Ultimaker/Cura #6756: cura mold setting

thelonious35, 2019-12-03:
hi please add mold number to the new version of cura

Ellecross, 2019-12-05:
uhm, what do you mean? Can you elaborate please? 
You can also fill up the New Feature template. :) 
Thanks

nallath, 2019-12-17:
Closing this due to inactivity.

closed 2019-12-17 by nallath

thelonious35, 2019-12-28:
need more wall number (mold setting) ı dont know how to speak english sorry
[image: cce.jpg]

Jaime van Kessel <notifications@github.com>, 17 Ara 2019 Sal, 16:56
tarihinde şunu yazdı:

Ghostkeeper, 2019-12-31:
You can adjust the thickness of the mould using the [Minimal Mold Width](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/blackmagic/mold_width.md) setting.

The mould itself is just printed using the normal print settings so if you need it to have more walls you can increase the [Wall Thickness](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/wall_thickness.md) setting.