Ultimaker/Cura #6757: Auto selection of material tab based on chosen extruder

68AndersK, 2019-12-04:
Often I spend time in Cura changing settings just to see it doesnt affect anything after slicing is done. Just because I havent selected the right material tab.

I would like Cura to automatically switch to the correct tab when a model is selected, based on what extruder chosen for that partcular model.

I believe this will help many new users, including myself. Have seen a few forum posts people asking why settings isnt accounted for when slicing just because of this. 

[image]

[image]

Ellecross, 2019-12-05:
Interesting @68AndersK!  
Let me just check with our @Yian-Lai. What do you think?

Ellecross, 2019-12-05:
We'll discuss this with the team!

Ghostkeeper, 2019-12-06:
@Ellecross I didn't see this in the CCB today though?

We need to think about what happens when you're making a dual extrusion print, too.

68AndersK, 2019-12-06:
Even more beneficial when doing dual extrusions.
It should only trigger upon model selection though. Of course you still need to be able to manually switch tabs like when using dial extrusions for  prints with only one model.

Thanks for considering it, really appreciated

Ellecross, 2019-12-09:
> @Ellecross I didn't see this in the CCB today though?
> 
> We need to think about what happens when you're making a dual extrusion print, too.

It's on the board now :P

Ellecross, 2019-12-10:
For reference: CURA-7035 @Yian-Lai