Ultimaker/Cura #6761: Plugin -

S2Gilly, 2019-12-04:
**Application version**
4.4

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

**Reproduction steps**
1. Something you did.
2. Something you did next.

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

Ellecross, 2019-12-05:
Hello @S2Gilly, is this similar to https://github.com/Ultimaker/Cura/issues/6730? ;)

S2Gilly, 2019-12-05:
I'm not 100% sure if it is similar, it very well could be related.  I'll double check to see if turning combing off works.  (I did do a search prior to posting and having my fat thumbs hit post before anything was actually in the post haha. But I guess I didn't search enough!! sorry! )

Also, I actually just went through the settings in cura again... and realized I forgot about the "retract at layer change" option.  I will be trying this this morning to see if that helps!

So I have several things to try.   

Thanks!!!

Ghostkeeper, 2019-12-06:
> (I did do a search prior to posting and having my fat thumbs hit post before anything was actually in the post haha. But I guess I didn't search enough!! sorry! )

Maybe that search didn't work because the other issue is written in German...

I'm closing this because to me this issue looks exactly alike #6730.

closed 2019-12-06 by Ghostkeeper