Ultimaker/Cura #6763: Cura 4.4.0 crashes at startup

Lamoresk, 2019-12-06:
Cura 4.4.0
Windows 10 18362.476 x64
RTX2070 

Printer : Creality CR-10

Since Cura 4.4.0, I get a fatal error when the main window opens. I already tried to uninstall, delete every cura directory and reinstall.

Traceback (most recent call last):
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\Machines\Models\IntentCategoryModel.py", line 98, in _update
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\Settings\IntentManager.py", line 119, in currentAvailableIntentCategories
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\Settings\IntentManager.py", line 61, in intentCategories
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\Settings\IntentManager.py", line 42, in intentMetadatas
KeyError: 'empty'

Thread 0x00004580 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thanks,
Matthieu

smartavionics, 2019-12-06:
This is a known regression which should be fixed in the next release. In the meantime you could install one of my releases from https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0 which have this fixed. Please read the README.md file there.

Ghostkeeper, 2019-12-06:
Yup, we fixed this one. Jury's still out on a version 4.4.1.

closed 2019-12-06 by Ghostkeeper

Lamoresk, 2019-12-06:
Thanks to both of you :)