Ultimaker/Cura #6764: Use of adaptatve layers introduces gaps in the supports

plgarcia, 2019-12-06:
Adaptative layers features induces interruption in supports, what makes this fantastic feature unusable when this problem occurs.

**Application version**

4.4.0 (The latest available as free download at the time, but the problem is not new and is existing in previous version)

**Platform**
Windows 7 54 Bits

**Printer**
CR-10

**Reproduction steps**
1 - Load Venus model https://www.thingiverse.com/thing:14715 or any model that needs support on the upper areas of the model)
2 - Activate adaptative layers
3 - Slice the model

**Actual results**
The support for the head is interrupted in its middle.

**Expected results**
The support should not be interrupted.

**Additional information**
screen shot
project file
one model showing this problem :  https://www.thingiverse.com/thing:14715

Adaptative layers is  very great feature that :
- Reduces significantly the need for supports (and material used)
- Improves significantly the quality of the model
- Reduces the time printing also for some models 

Note that this problem also exists with “tree support”

Find files here 
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/3931032/cura.log)
[Venus.3mf.zip](https://github.com/Ultimaker/Cura/files/3931043/Venus.3mf.zip)

a extra link dropbox if file not available in git
https://www.dropbox.com/s/02gukn27vq84eye/bugadaptativelayers.zip?dl=0

smartavionics, 2019-12-06:
Reduce the minimum support area and the gap goes away...

[image]

plgarcia, 2019-12-06:
Thank you for this answer.
It is not a bug sorry for the disturbance.

closed 2019-12-06 by plgarcia