Ultimaker/Cura #6766: Can't slice a model after mirroring it

alkhimey, 2019-12-06:
**Application version**
4.4.0

**Platform**
Ubuntu

**Printer**
Creality-Ender-3

**Reproduction steps**

1. Download the STL from here: https://cults3d.com/en/3d-model/art/dolphin-pendant
2. Open Cura.
3. Extract contents. Drag and drop dolphin_pendant.stl into Cura.
4. Press M (Mirror) and then click on one of the arrows on the model to mirror it.
5. Click on Slice 
6. You will see an error message: 
[image]
7. Go to file -> export. Select ASCII STL in the file type options.
8. You will get an error when exporting: 
 [image]
9. Press Ctrl-Z to revert the mirror.
10. Click on Slice - Cura now will not slice even the original model. 

I am not sure if this happens with any model. I tired this also with only one additional model and also got errors.

**Actual results**
Errors.

**Expected results**
Cura is a ble to slice the model after mirroring.

**Additional information**
None

alkhimey, 2019-12-06:
Tried to export to 3mf format. The export finished. When opening the exported file Cura crashed:
```
$ ./Ultimaker_Cura-4.4.0.AppImage 
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1065353216) >= this->size() (which is 841068)
Aborted (core dumped)
```

Ellecross, 2019-12-09:
Alright, I'm checking this out

Ellecross, 2019-12-09:
I have no problems with slicing this. I load the STL and mirror and slice, and it's fine. 

Can you slice it as you load the model? 

What settings have you changed? You can send the project file (File >Save) and then I'll check it out.

Ellecross, 2019-12-09:
Also, I love this ticket already because you filled up the bug template properly. :D :D :D :D :D :D :D <3

alkhimey, 2019-12-10:
Here is a project file after mirroring (the corrupted state which does not slice)

[CE3_FilamentGuideToughCleaner2longercuptop3.zip](https://github.com/Ultimaker/Cura/files/3947155/CE3_FilamentGuideToughCleaner2longercuptop3.zip)

I used a different model here.

alkhimey, 2019-12-10:
Might be relevant, I downloaded the `AppImage` from Cura's website and did not install it with `apt`.

christianpatrickgalles, 2019-12-11:
Having this issue as well

Ghostkeeper, 2019-12-11:
Yeah that project file indeed makes Cura crash for me as well! Same out of range error as in the original bug report.

rburema, 2019-12-13:
I can see why it crashes on load, the 3D data has become corrupted:

~~~
<triangles>
	<triangle v1="1065353216" v2="0" v3="1073741824" />
	<triangle v1="1082130432" v2="1077936128" v3="1084227584" />
	<triangle v1="1088421888" v2="1086324736" v3="1090519040" />
	.....
~~~

At least it isn't (also) a problem in libSavitar.

rburema, 2019-12-13:
Can't repro the original issue (mirror + slice, etc.) on Windows in either master-source or the 4.4.0 build :-/

smartavionics, 2019-12-13:
Is it a coincidence that those values are close(ish) to 0x40000000 ? Some kind of overflow/wrap around error?

rburema, 2019-12-13:
Probably not a coincidence, no :-) Over- or underflow is also the only thing I can think of as a cause... the question is; _where_ in the process?

alkhimey, 2019-12-13:
Additional information:

1. Ctrl-Z after mirroring does not fix the issue, slicing is not working even with a non mirrored model. But if I remove the model and add it again, slicing is working again.

2. Tried additional printer model, same issue.

3. Tried scaling, moving and rotating the model - had no issues with these.

If anyone else has this issue as well, there is an (obvious?) workaround for it. Load the model into Tinkercad online 3D modeling tool, mirror it and then export as STL. Cura will open it with no issues.

alkhimey, 2019-12-14:
Another workaround within Cura itself.
Download "Mesh Tools" plugin. After mirroring the model, use "Fix Model Normals" option on it.

nallath, 2019-12-16:
It seems to be an issue with how we package numpy, since i can't reproduce this when running it from source, but I can reproduce it when using the .appimage

Anyway, we're looking into this.

witem, 2019-12-22:
I have same problem on Arch with `.appimage`

FlailAway, 2019-12-23:
Me too. I tried 4.3.0 Appimage and all works as expected, Sliced fine in all Mirror directions, but not 4.4.0.

rburema, 2019-12-24:
@nallath Do we have an internal ticket nr. yet?

nallath, 2019-12-24:
Not sure, @LipuFei was looking into it. But so far we haven't made any sense out of it.

witem, 2019-12-24:
@nallath I can try add some additional logs if need

nallath, 2019-12-24:
We can reproduce it just fine, that's not really the issue. The problem is that if we run Cura from source with the exact same versions of numpy, it doesn't happen. If we run the appImage, we do have the issue (even though it's using the same pacakges...) .

FlailAway, 2019-12-24:
Just been messing with this. Not sure if it is any help, but. After a Mirror I tried selecting another lay-flat-face. It selected the opposite one that I clicked and put the face I clicked to the top. When I again tried lay-flat on the now-top face I had clicked before, it did nothing, it seems to think that face is already on the build plate.

FlailAway, 2019-12-24:
.p.s
I just tried with SmartAvionic's build 20191209 and same thing.

FlailAway, 2019-12-25:
Once the Mirror is done, it is impossible to Slice even if you undo the mirroring to the state it was when loaded where it would Slice OK.

Is it possibly a Permissions thing? A temporary file being created and not able to be deleted or something.

Ghostkeeper, 2019-12-30:
Just to be clear, the problem is in the mirror operation itself, not in the loading of the project or the slicing. The model file that was uploaded here contains coordinates that are about 1000km off the build plate. Cura is expected to slice only within the 1km area around the origin.

After mirroring and slicing the error in the Cura log is this:

```
2019-12-30 11:12:47,118 - ERROR - [Thread-3] UM.Logger.logException [84]: Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x875f1b8> caused an exception
2019-12-30 11:12:47,121 - ERROR - [Thread-3] UM.Logger.logException [88]: Traceback (most recent call last):
2019-12-30 11:12:47,124 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/build/env/4.4/inst/lib/python3.5/site-packages/numpy/core/fromnumeric.py", line 51, in _wrapfunc
2019-12-30 11:12:47,126 - ERROR - [Thread-3] UM.Logger.logException [88]: TypeError: Cannot cast array data from dtype('float32') to dtype('int64') according to the rule 'safe'
2019-12-30 11:12:47,129 - ERROR - [Thread-3] UM.Logger.logException [88]: 
2019-12-30 11:12:47,132 - ERROR - [Thread-3] UM.Logger.logException [88]: During handling of the above exception, another exception occurred:
2019-12-30 11:12:47,135 - ERROR - [Thread-3] UM.Logger.logException [88]: 
2019-12-30 11:12:47,137 - ERROR - [Thread-3] UM.Logger.logException [88]: Traceback (most recent call last):
2019-12-30 11:12:47,140 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 121, in run
2019-12-30 11:12:47,142 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/tmp/.mount_Cura-4cXQML1/usr/bin/plugins/plugins/CuraEngineBackend/StartSliceJob.py", line 306, in run
2019-12-30 11:12:47,145 - ERROR - [Thread-3] UM.Logger.logException [88]:     flat_verts = numpy.take(verts, indices.flatten(), axis=0)
2019-12-30 11:12:47,147 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/build/env/4.4/inst/lib/python3.5/site-packages/numpy/core/fromnumeric.py", line 181, in take
2019-12-30 11:12:47,150 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/build/env/4.4/inst/lib/python3.5/site-packages/numpy/core/fromnumeric.py", line 61, in _wrapfunc
2019-12-30 11:12:47,152 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/build/env/4.4/inst/lib/python3.5/site-packages/numpy/core/fromnumeric.py", line 41, in _wrapit
2019-12-30 11:12:47,155 - ERROR - [Thread-3] UM.Logger.logException [88]: TypeError: Cannot cast array data from dtype('float32') to dtype('int64') according to the rule 'safe'
```

And as Nallath said it doesn't reproduce for us when running from source on a developing environment. That indicates most likely that one of the involved dependencies has a different version, and it's been fixed in a more recent version, but we're not sure which. Nallath tried Numpy (since that is the library that does the actual mirroring computation for us) but apparently that was not the one.

dothebart, 2020-01-02:
Same here too. Its working with the 3.6.0 & 4.1.0 appimage,

michaellee8, 2020-01-05:
I got this issue as well, also because of mirroring.

gleonard3, 2020-01-17:
Also have this issue, running 4.4.1 on Chromebook's Linux using the AppImage.

nallath, 2020-01-21:
I'm not a 100% sure if I have a fix for this, but i did find something that could go wrong in this bit of code (and added a check to prevent it)

callmelightningjunior, 2020-01-22:
I'm getting this as well. I tried the suggested 'mesh tools' / 'fix normals' work around and it didn't work,

ariejan, 2020-01-29:
I got things to work with the following steps:

 1. Import the model STL.
 1. Mirror it
 1. Right click on the model and choose 'Reload all models'

Somehow this reload the model, but also mirrors it correctly.

waynert3d, 2020-02-10:
I am using rep-rap flavor, 4.4.1 app image and Cura on linux ( mint 19 cinnamon ). Will not slice after an item has been mirrored. I must clear the bed and reload items. All items slice and print well but will not slice if any item is reversed.

waynert3d, 2020-02-10:
I just tried ariejan's instructions and it works!

    Import the model STL.
    Mirror it
    Right click on the model and choose 'Reload all models'

poldy79, 2020-02-16:
Note: This workaround seems to work with stl, not with amf

Ghostkeeper, 2020-02-28:
This has been [reported as fixed](https://github.com/Ultimaker/Cura/issues/6861#issuecomment-590852119) in a different thread. Thanks for the debugging, guys!

closed 2020-02-28 by Ghostkeeper

Redsalamander, 2020-04-23:
Same issue today with 4.5 - I have 4.3 still installed. and used that it would slice and save as STL which would allow me to then slice it. 
Not a great work around but it does work.

AvanOsch, 2022-01-14:
Same issue today with 4.13.0 - AppImage
Installed Mesh Tools, which reported the model is not watertight.
Tried "Reload all models" workaround by @ariejan - Still working! (STL file)