Ultimaker/Cura #6769: Error Cura 4.3.0 Weedo

ese628, 2019-12-06:


ese628, 2019-12-06:
[image]

Ellecross, 2019-12-09:
I would really really like it if you: 

Give more description about the issue 
OR fill up the bug template 
OR attach a project file ;) 

This will help us reproduce your issue. 

Can you also try the same in [Cura 4.4](https://ultimaker.com/software/ultimaker-cura) please?

ese628, 2019-12-09:
Hi. I am trying to use the Ultimaker CURA version for the Weedo F192 printer, but it does not recognize the printer and does not save in X3G format. In version 4.4.0 this printer is not included.

ese628, 2019-12-09:
[image]

Ghostkeeper, 2019-12-09:
We don't publish any Cura version for Weedo F192. It appears that you've either downloaded an alternative version of Cura that we don't maintain, or you've installed a mod that got broken. Either way, there is little that we can do about it.

If you have a log file when trying to save that X3G file, you may report it in the issue list for the X3GWriter: https://github.com/Ghostkeeper/X3GWriter/issues
That is not something that Cura can do anything about either (but it's my own problem since I maintain that X3GWriter plug-in).

closed 2019-12-09 by Ghostkeeper

ese628, 2019-12-10:
Okay. Thank you!!

Ferjapolis, 2020-11-01:
Solo debes iniciar a Cura como administrador.