Ultimaker/Cura #6770: Unchecking "Uniform Scaling" doesn't apply setting

acsdog, 2019-12-07:
**Application version**
Cura 4.4.0

**Platform**
Windows 10 Home 64-bit
Intel i7-8700k
Nvidia GeForce GTX 1070 Ti

**Printer**
PrintrBot Play

**Reproduction steps**
1. Load model into cura
2. Select the model and open the scaling settings
3. Uncheck "Uniform Scaling"
4. Change one or two dimensions

**Actual results**
Changing the scale of two dimensions (e.g., x and y to 150%), with the uniform scaling option unchecked, the Z axis will automatically scale to the same percentage as x and y. At this point, the scale of each axis will change unpredictably (and sometimes without any user input whatsoever).

**Expected results**
The two axes who's scales are changed should not alter the third axis' scale.

**Project file**
[PP_Spiral_Box_Base_-_Blank.zip](https://github.com/Ultimaker/Cura/files/3934782/PP_Spiral_Box_Base_-_Blank.zip)

Ellecross, 2019-12-09:
Ahh yes. Scaling issues. The nemesis of @nallath and @Ghostkeeper 
We have an existing ticket about this at the moment @acsdog. Other axes change when you rotate the object, actually. 

For Cura UM devs, see ticket CURA-5630

Ghostkeeper, 2019-12-09:
The reproduction steps reported here don't reproduce, but yes if you've rotated the model before executing those steps then it reproduces.

This is a duplicate of https://github.com/Ultimaker/Cura/issues/4136.

closed 2019-12-09 by Ghostkeeper