Ultimaker/Cura #6771: intersectsPolygon function crash

dgiebert, 2019-12-07:
**Application version**
Cura 4.4.0
CuraEngine 4.4.0

**Platform**
Arch Linux
Qt 5.13.2
PyQt 5.13.2
Python 3.8

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Manually update the PKGBUILDs for cura, curaengine, uranium and arcus
2. Start Cura
3. Load an STL file (Should crash with every STL)

**Actual results**
Crash with an error message
```
**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
Provide a project which clearly shows the bug.
Rather than providing a model file, provide a project (.3mf or .curaproject) file. This will include all the settings you used, which is crucial to reproducing your problem.
To upload a project, try changing the extension to e.g. bug_example.3mf.zip so that GitHub accepts uploading the file.
If the file is too big, we recommend http://wetransfer.com, but other file hosts like Google Drive or Dropbox work well too. Please make sure the file is publicly available ie. "shared with everyone"

**Log file**
It is important to provide a log file. It might provide us info about why a certain material was selected, or is not compatible, for example.
Information about how to find the log file can be found at https://github.com/Ultimaker/Cura#logging-issues 

**Additional information**
(Extra information relevant to the issue)

Ghostkeeper, 2019-12-09:
Unable to reproduce this error here, running Cura, Uranium, CuraEngine and libArcus from source on latest master. I wonder what your set-up has differently. Perhaps it's only happening in Python 3.8? I'm still on 3.7. Can't imagine that would make a difference like this though. Perhaps Arch is a bit behind and ended up in a weird state when this was broken?

The expression `polygon_union.area == 0` seems correct to me since this is meant as an edge case handling if we create a polygon that has no area but still has vertices (like a single point). Perhaps such a check needs to be done in the intersection code as well?

tsipizic, 2019-12-09:
I get the same error with Cura 4.3.0 and arch linux

jukkaaakko, 2019-12-11:
Arch linux, Cura 4.3.0 and python 3.8. Same problem:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/cura/PlatformPhysics.py", line 53, in _onChangeTimerFinished
    build_volume.updateNodeBoundaryCheck()
  File "/usr/lib/python3.8/site-packages/cura/BuildVolume.py", line 262, in updateNodeBoundaryCheck
    if node.collidesWithAreas(self.getDisallowedAreas()):
  File "/usr/lib/python3.8/site-packages/cura/Scene/CuraSceneNode.py", line 98, in collidesWithAreas
    overlap = convex_hull.intersectsPolygon(area)
  File "/usr/lib/python3.8/site-packages/UM/Math/Polygon.py", line 184, in intersectsPolygon
    ret_size = (polygon_intersection.bounds[2] - polygon_intersection.bounds[0],
IndexError: tuple index out of range

rephlex, 2019-12-12:
Same issue here
Cura 4.4-PPA
Qt 5.12.5
PyQt 5.12.3

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/PlatformPhysics.py", line 53, in _onChangeTimerFinished
    build_volume.updateNodeBoundaryCheck()
  File "/usr/lib/python3/dist-packages/cura/BuildVolume.py", line 258, in updateNodeBoundaryCheck
    if node.collidesWithAreas(self.getDisallowedAreas()):
  File "/usr/lib/python3/dist-packages/cura/Scene/CuraSceneNode.py", line 98, in collidesWithAreas
    overlap = convex_hull.intersectsPolygon(area)
  File "/usr/lib/python3/dist-packages/UM/Math/Polygon.py", line 187, in intersectsPolygon
    ret_size = (polygon_intersection.bounds[2] - polygon_intersection.bounds[0],
IndexError: tuple index out of range

Thread 0x00007fb60f7fe700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb60ffff700 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb61c97c700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1875 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1406 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb61d17d700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb61d97e700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1875 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1234 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb61e17f700 (most recent call first):
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1155 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb635ffb700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 150 in _storeStderrToLogThread
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb6367fc700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 138 in _storeOutputToLogThread
  File "/usr/lib/python3.7/threading.py", line 870 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb637fff700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
[... 108 lines cut ...]

Thread 0x00007fb6a4d85700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007fb6a7586700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 890 in _bootstrap

Current thread 0x00007fb6be254740 (most recent call first):
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 306 in _logInfoWidget
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 156 in _createDialog
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 77 in __init__
  File "/usr/bin/cura.real", line 111 in exceptHook
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 342 in exec_
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 804 in run
  File "/usr/bin/cura.real", line 180 in <module>

rephlex, 2019-12-12:
> Tracing this down in UM.Math.Polygon shows that the polygon_intersection conditional does not work correctly in line 182.
> In line 161 this expression is used: `if polygon_union.area == 0:` which probably should also be adapted to `if polygon_union.area > 0:` in line 182 (as this also resolves the crash).

Can you provide a patch, i could test that here.

dgiebert, 2019-12-12:
```
--- Polygon.py	2019-12-12 16:38:55.083688740 +0100
+++ Polygon.py	2019-12-12 16:39:53.600398137 +0100
@@ -177,10 +177,9 @@
             return None
         if not (polygon_me.is_valid and polygon_other.is_valid):  # If not valid
             return None
-
         polygon_intersection = polygon_me.intersection(polygon_other)
         ret_size = None
-        if polygon_intersection:
+        if polygon_intersection.area > 0:
             ret_size = (polygon_intersection.bounds[2] - polygon_intersection.bounds[0],
                         polygon_intersection.bounds[3] - polygon_intersection.bounds[1],
                         )
```

Hope this helps

rephlex, 2019-12-12:
> ```
> --- Polygon.py	2019-12-12 16:38:55.083688740 +0100
> +++ Polygon.py	2019-12-12 16:39:53.600398137 +0100
> @@ -177,10 +177,9 @@
>              return None
>          if not (polygon_me.is_valid and [...]

I can confirm that patching /usr/lib/python3/dist-packages/UM/Math/Polygon.py fixes the issue for cura-4.4~201910291625~rev4146~pkg325~ubuntu20.04.1

Fun fact: The AppImage version of the 4.4 release does NOT throw the error in the first place ;)

Ghostkeeper, 2019-12-13:
Also running from source doesn't throw the error.

I'm worried that with that patch it could crash if `polygon_me.intersection` returns `None`, so be sure to also check for that.

alkhimey, 2019-12-13:
Additional information:

1. Ctrl-Z after mirroring does not fix the issue, slicing is not working even with a non mirrored model. But if I remove the model and add it again, slicing is working again.

2. Tried additional printer model, same issue.

3. Tried scaling, moving and rotating the model - had no issues with these.

If anyone else has this issue as well, there is an (obvious?) workaround for it. Load the model into Tinkercad online 3D modeling tool, mirror it and then export as STL. Cura will open it with no issues.

RockyTV, 2019-12-15:
I got the same issue even after updating to `cura=4.4.0-1` on Arch.

```
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/cura/BuildVolume.py", line 684, in _onSettingChangeTimerFinished
    self.rebuild()
  File "/usr/lib/python3.8/site-packages/cura/BuildVolume.py", line 536, in rebuild
    self.updateNodeBoundaryCheck()
  File "/usr/lib/python3.8/site-packages/cura/BuildVolume.py", line 258, in updateNodeBoundaryCheck
    if node.collidesWithAreas(self.getDisallowedAreas()):
  File "/usr/lib/python3.8/site-packages/cura/Scene/CuraSceneNode.py", line 98, in collidesWithAreas
    overlap = convex_hull.intersectsPolygon(area)
  File "/usr/lib/python3.8/site-packages/UM/Math/Polygon.py", line 184, in intersectsPolygon
    ret_size = (polygon_intersection.bounds[2] - polygon_intersection.bounds[0],
IndexError: tuple index out of range

```

The AppImage version works, but the Arch package is broken since the update to Python 3.8.

@Ghostkeeper you can add a check for both, `if polygon_intersection and polygon_intersection.area > 0:`

ninovanhooff, 2019-12-16:
Since Cura will use python 3.8 eventually, fixing this is worthwile. @rephlex, could you create a pull request for this?

Ghostkeeper, 2019-12-16:
Nevermind that. I made the change to Uranium. Please try again with the latest sources. It should be fixed!

For me the behaviour is exactly the same on Python 3.7. I can't test with 3.8 right now (as I'm only using that on a computer at home). Honestly I think the problem is more related to the version of Geos you're using or the Python bindings for that, though.

closed 2019-12-16 by Ghostkeeper