Ultimaker/Cura #6772: Ability to define/control print order for One-at-a-time printing

SneakyTiki, 2019-12-07:
I'm baffled that it has been 10 full releases of Cura since the struggle began, but it would be _extremely_ helpful if users could control the order in which models were printed when using one-at-a-time printing. I'm sure that the 'start in the back right corner, move forward, then step over to the left' ordering that Cura uses works fantastically for the Ultimaker printers. But it doesn't make sense for a any of the machines I use. Especially when printing parts taller than my gantries.

In Cura 3.4, the order which models were loaded in defined the order they would be printed: the first model loaded in was the last model printed, last one loaded was first. This was not the most convenient way of controlling print order, but at least it was something. Having absolutely no control whatsoever in the versions since means I'm still stuck regularly using 3.4 for production despite wanting to use many features that have been included since. A quick Google search reveals that this behavior was a 'bug' of some sort, and has been squashed to my great dismay.

It seems to me that anyone who uses their machines for any production (hobby or professional) would benefit greatly from this. One-at-a-time/sequential printing is the most effective way of doing large runs of parts and having no control over it seems criminal. I'm not even going to get into its complete absence when utilizing more than one tool, forcing us to slice objects individually in the location they should be printed and then splicing gcode manually.

Please allow users to utilize their own brain, and at the very least print models in the order which makes sense for their machines. If all you do is add a drop-down menu for selecting print order number, that would do. Probably a better way
[image]
 would be through the custom 'per-model' settings, maybe with a print order entry box? Or what I think would be the best way, would be to just allow us to define the starting point and progression behavior of the algorithm that is already being used. What I mean is: in machine settings, just like you define where the origin is, and the shape of the bed, let users define the starting point for the scan, and the primary and secondary direction. I have attached a high-quality illustration.

I sincerely hope this feature is given some attention, as I can't imagine I'm the only one pained by its absence. Cura does workflow so well in general, but this is a massive pain point.

**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ellecross, 2019-12-09:
Sounds good @SneakyTiki I'll bring this up with the team :) Thank you for your passion in sharing feature requests like this. <3

Ghostkeeper, 2019-12-09:
> In Cura 3.4, the order which models were loaded in defined the order they would be printed: the first model loaded in was the last model printed, last one loaded was first. [...] A quick Google search reveals that this behavior was a 'bug' of some sort, and has been squashed to my great dismay.

This was indeed a bug. The order in which models are loaded shouldn't matter (as you say, it's very weird for the workflow). It was causing damage to printers because the print head was allowed to collide with the previously printed models. The disallowed areas are constructed together with an ordering such that this shouldn't be possible, but since the order was arbitrary it was still possible.

The current direction is not necessarily one of the possible configurations that you propose. It could end up printing objects in a spiral or any shape if the objects' positions allow it. The ordering is computed to *maximise the available horizontal space*. You can't place objects closer together than this because one of the sides of the print head will always hit the other objects. Turns out that even for maximum packing of the build plate with the closest possible distance between all objects, there is always an ordering available in which these objects can be printed. And Cura will automatically find that ordering. *There is no possible way in which a user could manually think of a better ordering...* Except for one:

The very last object in the ordering can in theory be taller than the gantry height without the gantry colliding with previously printed objects. Cura doesn't take this into account. This is the actual problem you're trying to solve: Cura should allow the last object in the ordering to be taller than the gantry height. However Cura automatically comes up with an ordering so even if the last object is allowed to be taller, it'd be unclear to the user which object that is. As far as I'm concerned, that's where the current problem lies.

Ghostkeeper, 2019-12-09:
Closing this as a duplicate of https://github.com/Ultimaker/Cura/issues/4835. Please continue the discussion there.

closed 2019-12-09 by Ghostkeeper