Ultimaker/Cura #6773: [4.4.0] if open project (.3mf-file) no profiles or settings get in

Heikoho, 2019-12-07:
**Application version**
Cura 4.4.0

**Platform**
Linux ubuntu LTS 18.4

**Printer**
Anycube 4max

**Reproduction steps**
open File xxxx.3mf and model will load but no settings/profiles.
I testet with some other 3mf-files. it will never load settings/profiles.
Only the model will loaded.
of course I have set in the cura settings "open as a project"

**Screenshot(s)**
no screenshot available

**Actual results**
nothing

**Expected results**
in Cura 4.3.0 I was able to load a 3mf file and the saved profile or settings were read in

**Project file**

**Log file**
na

**Additional information**
na

Ellecross, 2019-12-09:
Can you attach the 3mf file @Heikoho ? thanks

Heikoho, 2019-12-09:
Hi,
here the attached ZIP-File, enclose two 3mf-files as example.
[3mf-files.zip](https://github.com/Ultimaker/Cura/files/3941712/3mf-files.zip)

Ghostkeeper, 2019-12-10:
I'm not able to reproduce this bug. Those project files load fine. It gets me an Anycubic printer with some German custom profiles in it:
[image]
[image]

Heikoho, 2019-12-11:
Hello,
i will check this.
In my Cura Version 4.3.0 it will do the loading ok, but in Cura 4.4.0 it don't.
thanks
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Ultimaker/Cura/issues/6773?email_source=notifications\u0026email_token=ANLLSCLUGV4BDLVU2HXWFRDQX5FOZA5CNFSM4JXR7TSKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEGOLE3I#issuecomment-563917421",
"url": "https://github.com/Ultimaker/Cura/issues/6773?email_source=notifications\u0026email_token=ANLLSCLUGV4BDLVU2HXWFRDQX5FOZA5CNFSM4JXR7TSKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEGOLE3I#issuecomment-563917421",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Heikoho, 2019-12-17:
Hi,
i downloaded Cura 4.4.0 again and deleted the other File.
I don't know why, but now it's working.

You can delete this thread.
Thanks

closed 2019-12-17 by Heikoho

ltachev, 2020-01-04:
I am having the same issue, on Linux (Manjaro/Arch), with theOS version (4.4.0), or  AppImage both 4.4.0 and 4.4.1 . The 3mf loads without asking for opening as a project, just as a model and that's it. Attaching the example 3mf. 
[xyz.zip](https://github.com/Ultimaker/Cura/files/4021965/xyz.zip)

fieldOfView, 2020-01-04:
@ltachev check the "Default behavior when opening a project file" setting in the General preferences.

ltachev, 2020-01-04:
@fieldOfView it's set to 'Always ask me this' . But it doesn't ask. Changing it to "Always open as a Project' does not do any change - it still opens it as a model.

Ghostkeeper, 2020-01-13:
@Itachev your bug does reproduce for me in Cura 4.4, but when I load it in the current master version it actually correctly pops up with a dialogue. It seems that this problem has been [fixed](https://github.com/Ultimaker/Cura/commit/dcaa10e409f89326ecd848c0482c82964135a3ae#diff-82d1e4f6c89475ef8c9263dc17b129d2) for the upcoming release.