Ultimaker/Cura #6774: Add Bed Mesh Save Slot Input Field to Print Profile settings.

Careless, 2019-12-07:
I just realized that I usually have to change the bed Mesh Level save slot in my print settings when switching from three print platforms I use (PEI, Mirror Tile, PCB board)

Is it possible to add a feature to change the mesh levelling save slot number input field for the particular print so that if I know I am going to print on a certain build plate, I can associate that print with the proper mesh slot?

Ghostkeeper, 2019-12-09:
Uh, bed mesh level save slot? Cura has no feature such as that. It sounds like you're talking about firmware settings of a particular type of printer. Cura has no say over these. Maybe you need a plug-in to support that type of printer where you can switch out things like the automatic levelling settings or something, but Ultimaker will not develop it for you unless it's got to do with their printers.

You should be able to change the Machine Settings dialogue window. Here's my dialogue after I've resized it to be very tall:

[image]

I'm closing this issue because it applies to firmware settings of a 3rd party printer. That is out of scope for Cura. Ask a plug-in developer to build it for you, if you need it!

closed 2019-12-09 by Ghostkeeper

Careless, 2019-12-09:
Thanks for closing without discussing. Very welcoming.

What I was suggesting was an input field for G29 G-Code Slot Number, either in the material profile or in the settings side-bar. Some materials require different build plates, and those can be swapped out and their accompanying maps can be loaded by adding G29 code to load from a save slot in Marlin, Repetier, and Duet.

Has nothing to do with my printer specifically. It's a common Gcode command.
Currently I've placed it in my start G-Code like most do with K-constant for Linear Advance, but this should be material specific as well- and there is a plugin for that.

I'll message the author of that plugin since you don't seem to think it's a valuable feature.

I've just tried resizing the machine settings window in MacOS Catalina, and it does not allow me to stretch the window.

Ghostkeeper, 2019-12-10:
Sorry, I didn't know about any "mesh level save slot" since my printer doesn't support it. Without having a printer that supports it, there is no way that I can test the feature, so it's infeasible to develop it. That's the reason why I closed it. We can't build this for you; you need to build it yourself and it's probably a good candidate for a plug-in.