Ultimaker/Cura #6775: Zoom Toward Mouse Direction Not Working on 4.3 or 4.4 for macOS

edm00se, 2019-12-08:
**Application version**
4.4 (also 4.3)

**Platform**
macOS Catalina (10.15.1), also noted previously w/ 4.3 on macOS 10.14 Mojave.

**Printer**
Creality CR-10, also noted with Monoprice Select Mini v1.

**Reproduction steps**
Ensure "zoom towards mouse" is checked in settings, perform zoom in/out with object loaded. (I've tried toggling off, quitting, opening, re-enablibg)

**Screenshot(s)**
[image]

**Actual results**
The zoom in/out seems to focus on a point in the ~"upper left" side of things from the default angle of perspective. This happens regardless of my initial mouse position while zooming.

**Expected results**
Zoom in/out will be towards mouse position.

**Project file**
None.

**Log file**
None.

**Additional information**
This may be related to #6490. Also, I've noticed with the update to macOS Catalina, many apps need(ed) to update to request permission to certain aspects, file system, etc. This may require an accessibility permission to get mouse position on the screen (< me guessing at things). As mentioned earlier this did appear to occur for me on 4.3 on macOS 10.14 Mojave (the previous OS release), so the Catalina permissions increase may not be the primary culprit.

Ellecross, 2019-12-09:
Thanks @Yian-Lai for helping me reproduce this. 
Will bring this up with the team! 

Thanks for the report @edm00se !

Ghostkeeper, 2020-04-22:
We had also seen this issue on Linux. It's now also reported on Windows in #7537.

ScruffR, 2020-04-23:
Just to reiterate one point of my now closed duplicate: When fixing the bug can you also consider renaming the feature as one is not zooming toward the mouse *direction* but mouse *position*?

Ghostkeeper, 2021-03-11:
I wouldn't be opposed to that rename, no. It always felt to me like a grammar problem, like it should've read "zoom in the direction of the mouse".

seaniepie, 2021-03-11:
So, this was logged in Dec 2019 and still hasn't even been assigned yet. So Ultimaker is another one of 'those' companies. Good luck to us having this fixed like ever, let alone this century! I give up providing any kind of bug reporting to developers like this who are so busy creating new features(bugs) to fix the ones they already have. An ever perpetuating vortex of disappointment and growing unusability.

STekZeNIO, 2021-05-13:
On Cura v4.9.1, this issue is back yet again only this time with "Zoom Toward Mouse Direction" greyed out. 
How can you not surprised to see basic and essential feature that creates such inconvenience level — not being resolved but coming back again and again?

seaniepie, 2021-09-10:
6 months on and STILL not assigned to anyone! **_Still_** too busy creating new bugs to fix old ones.

sovtek, 2022-04-28:
Well, it's April 2022 and I still have this problem in version 4.13.1 on mac os catalina. The setting is enabled but the scroll zoom always just goes off into the same strange left direction when the setting is enabled.

seaniepie, 2022-04-28:
It’s also still an issue in v5.0beta on the latest MacOS. It doesn’t look like it will ever get fixed.

seaniepie, 2022-05-09:
I've posted a [pull request](https://github.com/Ultimaker/Uranium/pull/817) to hopefully fix this.

seaniepie, 2022-11-08:
Yep, all fixed (mostly). There is still a slight offset but it is certainly better than we ever had before. Thanks Eric.