Ultimaker/Cura #6776: Issues Updating to Cura 4.4 - profiles from 4.0 cannot be imported - Creawesome values are not getting applicated in 4.4

fakio87, 2019-12-08:
I installed Cura 4.4 (before using 4.0 with Creawesomemod with an CR-10)

=> it says that from version 4.2 Creasesomemod is "on board". I could add a cr-10 printer but cannot see the build plate and alle standard creawesome settings are not set.

I also tried to import my old profile with the adjusted settings. Import was successfull - but no values or wrong values are imported. So the imported profile is useless.

I am working with Cura for years but never saw a software that works so bad with updates.
Cura cannot be upgraded within Cura and one always has to make a fresh install. There is always struggle with importing the old settings. 

Even sometimes the new version notices there are old settings but always telling they were corrupt and need to reset.

Ellecross, 2019-12-09:
Can you PM me your config folder from version 4.2 and then walk me through with your scenario? 
I'd like to try to reproduce this behavior so we can investigate further. 

Something like this: 
1. load your 4.2 folder
2. open 4.4 
3. import profiles (from 4.2) into 4.4.

fakio87, 2019-12-12:
Hi,

Do you mean the folder in Application Support/Cura ?

Actually I did install 4.4 and import the exported profiles from 4.0 in 4.4 as there seems to be no Creawesome setting in 4.4 when adding a CR-10 printer.

I do not find the pm feature here ... how to PM

What files do you need exactly and is there no feature that will just give the option to load the settings from previous versions?
I know that that I can restore an backup in Cura but this will just tell me that all has to be reset as something is wrong with the settings.

Ghostkeeper, 2019-12-12:
> Do you mean the folder in Application Support/Cura ?

Yes. Alternatively, open up Cura and go to Help -> Show configuration folder. That's the folder Ellecross asked for.

There is no PM feature on Github. You'd do best to upload a .zip file to this issue thread.

fakio87, 2019-12-12:
Okay, here we go
[4.0.zip](https://github.com/Ultimaker/Cura/files/3957523/4.0.zip)
[4.4.zip](https://github.com/Ultimaker/Cura/files/3957524/4.4.zip)

Ghostkeeper, 2019-12-16:
Upgrading this folder to 4.4 gives the following error messages:

```
2019-12-16 09:32:45,983 - ERROR - [MainThread] UM.Settings.ContainerStack.deserialize [390]: When trying to deserialize creality_cr10_extruder_0 #3, we received an unknown container ID (cr10_0.4_PLA_standard)
2019-12-16 09:32:45,983 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Error in the format of container creality_cr10_extruder_0 #3: When trying to deserialize creality_cr10_extruder_0 #3, we received an unknown container ID (cr10_0.4_PLA_standard)
2019-12-16 09:32:45,983 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2019-12-16 09:32:45,984 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/Settings/ContainerRegistry.py", line 183, in findContainers
2019-12-16 09:32:45,984 - ERROR - [MainThread] UM.Logger.logException [88]:     new_container = provider.loadContainer(metadata["id"])
2019-12-16 09:32:45,984 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/plugins/LocalContainerProvider/LocalContainerProvider.py", line 68, in loadContainer
2019-12-16 09:32:45,984 - ERROR - [MainThread] UM.Logger.logException [88]:     container.deserialize(f.read(), file_path)
2019-12-16 09:32:45,984 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/Settings/ExtruderStack.py", line 165, in deserialize
2019-12-16 09:32:45,984 - ERROR - [MainThread] UM.Logger.logException [88]:     super().deserialize(contents, file_name)
2019-12-16 09:32:45,984 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/Settings/CuraContainerStack.py", line 259, in deserialize
2019-12-16 09:32:45,984 - ERROR - [MainThread] UM.Logger.logException [88]:     serialized = super().deserialize(serialized, file_name)
2019-12-16 09:32:45,985 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/Settings/ContainerStack.py", line 391, in deserialize
2019-12-16 09:32:45,985 - ERROR - [MainThread] UM.Logger.logException [88]:     raise ContainerFormatError("When trying to deserialize %s, we received an unknown container ID (%s)" % (self.getId(), container_id))
2019-12-16 09:32:45,985 - ERROR - [MainThread] UM.Logger.logException [88]: UM.Settings.ContainerFormatError.ContainerFormatError: When trying to deserialize creality_cr10_extruder_0 #3, we received an unknown container ID (cr10_0.4_PLA_standard)
2019-12-16 09:32:45,986 - ERROR - [MainThread] UM.Settings.ContainerStack.deserialize [390]: When trying to deserialize Creality CR-10 #2, we received an unknown container ID (cr10_global_standard)
2019-12-16 09:32:45,987 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Error in the format of container Creality CR-10 #2: When trying to deserialize Creality CR-10 #2, we received an unknown container ID (cr10_global_standard)
2019-12-16 09:32:45,987 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2019-12-16 09:32:45,987 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/Settings/ContainerRegistry.py", line 183, in findContainers
2019-12-16 09:32:45,987 - ERROR - [MainThread] UM.Logger.logException [88]:     new_container = provider.loadContainer(metadata["id"])
2019-12-16 09:32:45,987 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/plugins/LocalContainerProvider/LocalContainerProvider.py", line 68, in loadContainer
2019-12-16 09:32:45,987 - ERROR - [MainThread] UM.Logger.logException [88]:     container.deserialize(f.read(), file_path)
2019-12-16 09:32:45,988 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/Settings/CuraContainerStack.py", line 259, in deserialize
2019-12-16 09:32:45,988 - ERROR - [MainThread] UM.Logger.logException [88]:     serialized = super().deserialize(serialized, file_name)
2019-12-16 09:32:45,988 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/Settings/ContainerStack.py", line 391, in deserialize
2019-12-16 09:32:45,988 - ERROR - [MainThread] UM.Logger.logException [88]:     raise ContainerFormatError("When trying to deserialize %s, we received an unknown container ID (%s)" % (self.getId(), container_id))
2019-12-16 09:32:45,989 - ERROR - [MainThread] UM.Logger.logException [88]: UM.Settings.ContainerFormatError.ContainerFormatError: When trying to deserialize Creality CR-10 #2, we received an unknown container ID (cr10_global_standard)
```

And this is correct. Because neither the [CreawsomeMod sources](https://github.com/trouch/CreawsomeMod/tree/beta/4/src/quality) nor the [Ultimaker sources](https://github.com/Ultimaker/Cura/tree/4.0/resources/quality) at the time contained any quality profile called `cr10_global_standard` or `cr10_0.4_PLA_standard`

Looking at this further, if I open up Cura 4.0 with your 4.0 folder, I get the same error messages.

I think your 4.0 installation was in a weird state with different mods installed. We tried accommodating users of the CreawsomeMod since it was installed by a significant user base, but you seem to have something else installed as well.

closed 2020-06-19 by no-response[bot]