Ultimaker/Cura #6777: 4.3.0 Fails to load new quality resources from \resources\quality

photodude, 2019-12-08:
**Application version**
Cura 4.3.0

**Platform**
Win 10 Version	10.0.18362 Build 18362
Processor	i7-4770
DDR3 RAM	32.0 GB
Graphics Nvidia GTX 1070

**Printer**
BIBO2 E1 Single

**Reproduction steps**
1. create new global Quality resource configuration file super.inst.cfg
2. attempt to import profile with "super" quality level

**error messages**
Failed to import profile from /chep_cura_profiles_4.3/CHEP_Cura_Profiles_4.3/Magic0.12.curaprofile: Could not find a quality type super for the current configuration.

**Actual results**
fails to import profile as new global quality_type = super for fdmprinter based printers didn't load/register with cura

**Expected results**
new global quality_type = super for fdmprinter based printers should load/register with cura allowing the import of profiles which are defined as needing quality type super

**Additional information**
possibly related to 
https://github.com/Ultimaker/Cura/issues/6453
https://github.com/Ultimaker/Cura/issues/6552

new super quality resource file
[super.inst.cfg.zip](https://github.com/Ultimaker/Cura/files/3937027/super.inst.cfg.zip)

Profile version tested with import
[Magic0.12.curaprofile.zip](https://github.com/Ultimaker/Cura/files/3937031/Magic0.12.curaprofile.zip)

Ghostkeeper, 2019-12-09:
Please try this with Cura 4.4. We've refactored the entire system that determines which profiles are available so it's likely that this sort of bug would've been filtered out. I agree that it should find the global quality profile `super` in your case.

photodude, 2019-12-09:
@Ghostkeeper I have retried in 4.4 with the same results.
I did notice that my printer definitions did not have the following line
```json
"has_machine_quality": true,
```
I tried adding that line but that didn't fix the profile import when global quality profile `super` was needed. Additionally adding that line to my printer profile caused all of my other profiles to disappear from the list leaving only the default cura profiles. My print profiles repopulated in the list once the json line above was removed from the printer definition.

photodude, 2019-12-09:
@Ghostkeeper I retried again in 4.4 and the import worked. Not sure why it failed the first time I tried.

Ghostkeeper, 2019-12-10:
If `has_machine_quality` is set to `True`, I'd expect that it will *not* load the global quality profiles any more. Just the profiles that are specific to the printer. That's the aim of that metadata setting, anyway. So better leave it at `False` indeed.

Ghostkeeper, 2019-12-10:
Maybe something went wrong the first time. Did you have Cura open while you were modifying things?

From my point of view, it looks like this bug was luckily fixed in the refactor of 4.4...

photodude, 2019-12-10:
@Ghostkeeper I don't remember if cura 4.4 was open, but it's possible it was open when I moved the new global quality setting over to the 4.4 folder and I forgot to reload.

I'm going to proceed with my testing of this new quality setting as though things are working as expected in 4.4

Good to know that `has_machine_quality` is just for if there are quality settings for a specific printer (something I may consider working on later)

Shall we close as fixed in 4.4 although this is a 4.3 specific issue?

photodude, 2019-12-11:
@Ghostkeeper I just remembered that before retesting I had changed the metadata `weight` from `-1` to `2` as I noticed the other quality settings had unique values for the metadata `weight` field.

Ghostkeeper, 2019-12-11:
I don't expect that changing the weight would break anything. That value is used to determine the order in which these profiles should appear in the list, nothing else. So whether it took the old weight or the new one shouldn't really make much of a difference apart from the ordering in which the profiles appear in the list.

Ghostkeeper, 2019-12-11:
I'll close this as fixed in 4.4 then. I think we (developers) got lucky that we fixed this but you got unlucky that it went wrong with modifying files while Cura was open. It's easy to make a mistake when modifying those files manually.

closed 2019-12-11 by Ghostkeeper

photodude, 2019-12-12:
@Ghostkeeper if `weight` just determines the order in which these "quality levels" are shown in; is it possible that a conflict happens if two quality levels have the same `weight` value?

Ghostkeeper, 2019-12-13:
Unless we have a bug, I'd expect the order to then be arbitrary. This is the place where it's being used:

https://github.com/Ultimaker/Uranium/blob/6d669eb2b64c8fd5beec05ab238a3b11ad7840b8/UM/Settings/InstanceContainer.py#L688-L697

If the weights are equal, `__lt__` will always return False in which case the ListModels should resolve it arbitrarily.