Ultimaker/Cura #6779: Add support for XYZPrinting printers

jpfarias, 2019-12-09:
**Is your feature request related to a problem? Please describe.**
Not related to a problem

**Describe the solution you'd like**
Being able to print to my XYZPrinting Da Vinci Pro 1.0 directly from Cura, eliminating the need of exporting to gcode then loading on XYZWare to print.

**Describe alternatives you've considered**
The only alternative so far is to save the gcode to a file and print it from XYZWare

**Affected users and/or printers**
XYZPrinting Da Vinci Jr, Pro 1.0, 3-in-1 and some other printers from the same manufacturer

**Additional context**

There is already an open source app that can print to XYZPrinting printers, the code can be found here: https://github.com/reality-boy/miniMover

I don't know how difficult it is to add support for new printers to Cura, but hopefully this won't be too big of a task since there is already some code that accomplishes that on the link above.

Ellecross, 2019-12-09:
Thanks for the request @jpfarias but Ultimaker never adds new machine definitions ourselves. All of the non-ultimaker printers in Cura now were all contributions from the community. :) 

You can make it yourself - check out @LipuFei 's guide on [how to add Machine Profiles](https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura), then you can create a pull request. I know PRs have a reputation that they are processed slow - but maybe not for adding printers. ;) 

Good luck!

Ghostkeeper, 2019-12-09:
Send me a printer and I'll spend a few hours of my free time to make a profile for you as payment :wink: But I'll close this because there is no way in which we can create profiles without having the actual printer.

closed 2019-12-09 by Ghostkeeper

ChrisTerBeke, 2019-12-09:
I know from personal experience that the printers from XYZ have proprietary drivers and do not run standard Marlin-compatible G-code. Chances of getting this to work, even with an actual printer, are very slim. Also all of their printers have slightly different header encoding to make it even more difficult to use 3rd party driver software (probably on purpose). Creating decent print profiles is probably the least of your worries :P

jpfarias, 2019-12-09:
Yeah, I know... The way I am working with it now is I load my models on Cura with my custom settings for the printer then I save the gcode to a file. Once I have the gcode I can send it to the printer with XYZWare Pro, XYZPrint or with that little app (minimover) who seems to have cracked their usb protocol of how to upload gcode to the printer.

I was hoping there would be a way to integrate the code from minimover into Cura so I could send the gcode directly from Cura to the printer and avoid always having to export from Cura then open on another app and only then send it to printer. It would basically remove my need to have XYZ software on my PC.

For now I guess I will keep doing Cura -> GCode -> MiniMover

ChrisTerBeke, 2019-12-09:
If MiniMover works 'out of the box', probably the easiest implementation would be to call that binary from Python in a Cura plugin. The other option is to re-implement the protocol stuff that MiniMover hacks into the G-code in Python so it can be added to Cura as plugin as well.

jpfarias, 2019-12-09:
The communication with the binary might actually be a good starting point. I took a peek at the code too and it doesn't seem too complicated. 

As of communication with the binary, on Linux at least the way it works is you give it the gcode and it converts that to a `.3w` file which you can then call the binary again passing the `.3w` file to send it to the printer.

They also have a UI for windows only though where you can just give it a gcode file and it converts and sends to the printer in one step, shows a progress bar and all. It can also get some printer monitoring information such as extruder temperature, bed temperature and more.