Ultimaker/Cura #6780: Can`t change machine settings. Cura 4.4

Nemori27, 2019-12-09:
**Application version**
Cura 4.4

**Platform**
Windows 10

**Printer**
Tevo Tarantula

**Reproduction steps**
1. Go to Settings > printer > mannage printers > machine settings and change nozzle diameter fron 0.4 to 0.2
2. Apply changes, tried with Enter button, close, or Escape button. Every time i reopen settings there is still 0.4 mm nozzle. 

**Screenshot(s)**
[image]
[image]

**Actual results**
Nothing happens, every time i reopen settings there are still 0.4 mm nozzle

**Expected results**
0.2 mm nozzle in extruder parameters

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/3939752/cura.log)

**Additional information**
Had no such problem in earlier versions

rburema, 2019-12-10:
What's your locale set to?

Nemori27, 2019-12-10:
> What's your locale set to?
Not shure that i understand you, can you formulate question it another way?

rburema, 2019-12-10:
I asked that because I think it might be a duplicate of #6685. You might want to have a look there as well, to see if it helps you.

Like I say there:
> A 'locale' in the sense that I meant is basically the (collection of) internationalisation settings that (a program or) the OS is set to. For instance, if I set my locale to Dutch (nl_NL), then, not only will the language change, but the default decimal separator will be a `,` while the thousands separator will be a `.` (the opposite of what it is in English (en_EN)).

Ghostkeeper, 2019-12-10:
So basically, what's the language settings of Windows? This includes:
* The language of the operating system.
* The localisation settings (so whether to display time as `MM-DD-YYYY` or `DD-MM-YYYY` and such).
* The keyboard layout.

In particular the second one is probably the one that causes this, as it affects how to display numbers (as `123.456,00` or `123,456.00`).

Nemori27, 2019-12-10:
> So basically, what's the language settings of Windows? This includes:
>
> * The language of the operating system.
> * The localisation settings (so whether to display time as `MM-DD-YYYY` or `DD-MM-YYYY` [...]

OS language is russian, keyboard layout english, date DD-MM-YYYY

Ghostkeeper, 2019-12-10:
Yeah it looks like this is a duplicate of #6685 then. Please continue discussion/investigation there.

closed 2019-12-10 by Ghostkeeper

Nemori27, 2019-12-10:
> Yeah it looks like this is a duplicate of #6685 then. Please continue discussion/investigation there.

Yep, guess youre right, situation seems the same, thank you