Ultimaker/Cura #6782: [4.4.0] Adaptive layers options unexpectedly hidden

doldroyd, 2019-12-09:
**Application version**
4.4.0

**Platform**
Tested on Windows and Mac OS

**Printer**
Ender 3

**Reproduction steps**
1. Select "Dynamic Quality" default profile
2. Select "Show all settings" under customs setting menu
3. Expect to see "Adaptive Layers Maximum Variation", "Adaptive Layers Variation Step Size", and "Adaptive Layers Topography size" options.

**Actual results**
None of the three options listed are visible, despite "Show all settings" being selected and them being checked as visible in the settings menu.
If "Use Adaptive Layers" is manually deselected and reselected, the settings show up.

**Expected results**
Adaptive Layer settings should be visible if "Use Adaptive Layers" is selected by default in the current profile.

rburema, 2019-12-10:
Can confirm this happens.

Whatever should happen, at least the behaviour should not be different between the default checked and the user selected checked (in this case).

MalHaak, 2022-12-21:
This is still an issue. I'm on 5.2.1 under Linux
Doesn't matter if I use a fresh, new project or not.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

closed 2024-12-07 by github-actions[bot]

Carnarts, 2025-07-05:
This still isn't working for me, adaptive layer options are hidden even when checked to be visible in the settings