Ultimaker/Cura #6786: Blank window (transparent) on Fedora 31 with Wayland

gferon, 2019-12-10:
**Application version**
4.1.0 and 4.4.0

**Platform**
Fedora 31, Mesa DRI Intel(R) HD Graphics (Coffeelake 3x8 GT2)

**Printer**
N/A

**Reproduction steps**
Just launch cura, when exporting `XDG_DESKTOP_TYPE=x11` cura opens fine.

**Screenshot(s)**
[image]

**Actual results**
Empty window

**Expected results**
Cura should open

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
Nothing relevant from diffing cura forced to run on X11 and running on Wayland

**Additional information**
I'm not entirely sure, but it looks like the issue lies around the way the OpenGL context is initialized. When removing the lines that detect the OpenGL 4.1 version and forcing to run it with OpenGL 2.0, cura opens and runs fine under wayland. I haven't been able to figure out which flag could cause a problem in Uranium's `OpenGLContext.py`.

Related bug on Fedora: https://bugzilla.redhat.com/show_bug.cgi?id=1774806

rburema, 2019-12-11:
Does the AppImage version work at least? The PPA's are up to the individual maintainers.

hroncok, 2019-12-11:
What PPAs? This is the Fedora regular package from the distro, no personal package. We are the maintainers and we opened this to talk to you about it.

We can try the appimage, but doesn't it force x11?

rburema, 2019-12-11:
My apologies.

I'm not on Linux at the moment, so I'll ask my colleagues.

hroncok, 2019-12-11:
appimage on wayland system shows the window.

rburema, 2019-12-11:
My colleagues suggest your Qt version needs to be build with Wayland support.

gferon, 2019-12-11:
> appimage on wayland system shows the window.

yeah, I already tried that, it's running with X because I guess the AppImage comes only with Qt built with X support.

gferon, 2019-12-11:
> My colleagues suggest your Qt version needs to be build with Wayland support.

That's already the case in Fedora for a little while now. The thing is, the loading pop-up shows fine, but the main window stays empty. The only thing I've been able to figure out is that by commenting out line 112 to line 168 of `OpenGLContext.py` in Uranium, the window renders fine and Cura works under Wayland.

Ghostkeeper, 2019-12-11:
If you're using OpenGL 2.0 that means that your layer view will look a bit dated, and the "lay flat on face" option in the rotate tool is not available. It's an option but sub-par indeed. Maybe Qt is not playing nicely with your video drivers?

erickj, 2019-12-13:
+1 for me on this issue.

I've been running fedora with wayland since fedora made it the default (sometime around 2016?). in fedora 30 cura worked without issue. after upgrading to fedora 31 I have the same problems described as the bug report

here is a gist that captures the output of `$cura` along with the rpm packages that list installed dependencies for cura on my machine:

https://gist.github.com/erickj/4e2e82da72de087f5c324b289c2d730d

please let me know if there is more useful information I can provide here

saintaardvark, 2019-12-19:
I'm also on Fedora 31 and having this issue.  If there's anything I can do to help, please let me know.

closed 2022-06-23 by gferon