Ultimaker/Cura #6788: Issue while updating from Cura 4.2.1 to Cura 4.4: All materials lost

AagamShah97, 2019-12-10:
**Application version**
4.4.0

**Platform**
Windows 10

**Printer**
Ultimaker 2 Extended

**Reproduction steps**
1. Updated from version 4.2.1 to version 4.4
2. Added parts for slicing the way I normally do

**Screenshot(s)**
[image]

**Actual results**
No material selection

**Expected results**
I should have received an option to select the material and hence receive an estimate of the mass to be printed

Ghostkeeper, 2019-12-11:
The Ultimaker 2 Extended doesn't support material profiles, so it's strange that you could add them in Cura 4.2.1. When I try this, I can't do that.

But perhaps you had an Ultimaker 2 Extended with the Olsson block extension installed? There could be something going wrong there in 4.4 due to a refactor we made. I'd expect that if you add a new printer called "Ultimaker 2 Extended with Olsson Block" then your old profiles are still there.

Solonari, 2019-12-14:
Whether the device supports profiles or not, this was a nice feature of earlier versions of Cura.
The basic model human knows how much they paid for that kilogram of filament, and it was useful to not just see "0g - 14.02m" and see the length of filament used, but the overall mass and projected cost of the print instead of "empty - 14.02m  0g  $0.00" in material estimation.
Please consider bringing back this feature.

AagamShah97, 2019-12-17:
@Ghostkeeper I can't remember installing the Olsson block extension while I was using Cura 4.2.1 for the Ultimaker 2 Extended (it's been very long since I have used Cura). I can try adding it now though. Thanks!

heppg, 2020-01-21:
Similiar effect in my environment.
- win10, 4.4.1 update, materials for ultimaler2 (was previously used) are lost. Other printers show materials. Deinstallation and new installation did not help.
- linux install; 4.4.1 fresh install for this machine, no materials available for multiple printers.

mahtDFR, 2020-01-24:
The Ultimaker 2 Extended doesn't support material profiles. If you change them it shouldn't have any effect on the printer. Are you able to share a zip of a project file from the old version so we can try and reproduce to see what's happening here?

fieldOfView, 2020-01-24:
The Ultimaker 2 family does support material profiles if you set the GCode flavor to 'Marlin'.

closed 2020-06-19 by no-response[bot]