Ultimaker/Cura #6790: Cura 4.4 Fonts Issue at Configure Setting Menu

Ywang1233, 2019-12-11:
**Application version**
Cura 4.4 Windows

**Platform**
Windows 10 Home 64bit 1909
AMD 2600X 3.6GHz
RX580 8GB
DDR4 3200 16GB 

**Printer**
Custom Corexy BLV Cube

**Reproduction steps**
1. Click Perfomance =>Configure Cura
2. Fonts in setting meun shown as squeezed and garbled.
**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
I have installed Noto fonts. It makes no difference.
I have tried the method by putting theme.json listed in https://github.com/Ultimaker/Cura/issues/4784#issuecomment-443146888 to setting folder. No difference.

Texts in menu still squeezed and garbled

**Expected results**
Clear and readable texts in setting menu. 
**Project file**
[Cura Theme.zip](https://github.com/Ultimaker/Cura/files/3948766/Cura.Theme.zip)

**Log file**
N/A

**Additional information**
Thank you!

Ghostkeeper, 2019-12-11:
Whoa, this is pretty seriously garbled. I haven't seen anything like this before except maybe the issue with the Cyrillic characters on Linux. I wonder why it's different on your computer than anyone else since the font rendering system is not something that's loaded from Cura's configuration or anything.

Ghostkeeper, 2019-12-11:
Is it still messed up in the Machine Settings dialogue?

The labels in the main preferences screen are drawn using system settings for anti-aliasing, I think.

Ywang1233, 2019-12-11:
If I roll the setting list to very bottom, the fonts would change back to normal. But once I roll it back when the Italic sentance' You will need restart the application for blablabla...' shows up, all the fonts are garbled again. I havenot seen any issue like this before either.  Here is the Mechine Setting menu, readable but not fully correct font size. 
[image]

Ywang1233, 2019-12-11:
Now I scaled my system display to 120%. The configure menu is readable and much better. I still can see the Italic font not fully correct displayed.
[image]

The mechine setting menu font remains same without noticable difference.

Ywang1233, 2019-12-11:
By the way, my display setting is 1920*1080 60Hz, previously with 100% scale, current scale is 120%.

Ellecross, 2019-12-16:
hey @Ywang1233 do you have specific regional system settings that are different than english? I cannot reproduce your issuem unfortunately. :/

Ywang1233, 2019-12-16:
Thanks for comment. Yes, I do have different setting.  But only one of two
my computers I installed cura on has this issue. Both of them are set as
region China and display language simplified Chinese. The computer
encountering this problem installed windows 10 home from genuine Chinese
image. The other PC with no issue are factory installed English windows 10
home from Dell. But even if I set cura with Chinese, I still see this
issue. So I am not very sure is it because my system setting.

Ghostkeeper, 2019-12-17:
Maybe your system font doesn't have symbols for Latin characters? What is your system UI font?

Ywang1233, 2019-12-17:
Well, I did install the Noto font set, so it may not be miss of Latin characters I think. Even I set Cura to simplified Chinese, I still met this issue. It might be more like a font size issue, not font itself. Now since I run Cura with 125% system UI scale, I do not see this issue anymore.

Yanda

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

Ghostkeeper, 2019-12-18:
Can you try something for me?

1. Download this file: http://dulek.net/work/theme.json (perhaps right click -> save as)
2. Save that file in place of `C:\Program Files\Ultimaker Cura 4.4\resources\themes\cura-light\theme.json` (or wherever you installed Cura). You'll need to provide admin access to do this, so download it first to somewhere like your desktop and then copy it over with the file browser.
3. Make sure your system UI scale is back at normal, where you originally had the issue.
4. Restart Cura. Are the fonts normal now?

Ywang1233, 2019-12-19:
Thanks for help, unfortunately, Issue remains. I recorded a short screen record video to demonstrate this issue, hope you can see it. I shared it with a google drive link cause it is over sized for attachment.
https://drive.google.com/open?id=12oK0UJcXczwGo69sVpPMOOVPuI05FB03

Yanda

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

Ghostkeeper, 2019-12-20:
Hrm, yeah I see it. That was the one idea I had that could've been causing this. With that other theme file it's drawing this text just in the OS default font size and style. It should be the most basic way to draw text, really.

It's very hard to debug this remotely. We're only guessing in the dark here until we know how to reproduce this problem.

Ywang1233, 2019-12-20:
Yeah, I understand. The other functions act really well. Thanks alot.

shmyu12, 2020-03-23:
I have same problem and I use RX580.
I guess AMD graphics card is the cause.
I don't know any more...

Ywang1233, 2020-03-23:
Yep, you are right. I have the same graphic card. AMD hasn't fixed the
driver yet.

AndreVanKammen, 2020-05-12:
[image]

I have the same Problem,
Windows 10
GTX 1070
4K TV as monitor

Mathieu1972, 2020-06-25:
Hi,
I had the same problem with Cura 4.6.1. It was just displayed unreadable characters in fields of parameters. You can see what is inside the field of parameter by copy/paste in a notepad to confirm that there is a parameter but it is still not readable.
The problem for me was the font Noto Sans that was not known or set correctly. Close Cura and then 
edit with Notepad++ or other editor the C:\Program Files\Ultimaker Cura 4.6\resources\themes\cura-light\theme.json and replace the Family font to "Times" instead of "Noto Sans". Times is a common type of font in most systems.

You can do the same for the others theme.json in the others directories.
Restart finally Cura and it should work well..

Ywang1233, 2020-06-26:
I tried this method first, it doesn't work for me. Seem like multiple
causes could lead to this phenomena.

eduncan911, 2020-07-17:
I'm getting the same.  Clean Windows 10 install, nothing installed.

However, I am trying to use Dremel's branded version of Cura, which is version... Actually, I have no clue as I can't read the About screen.

[image]

[image]

[image]

if it matters, I am running Windows 10 within VirtualBox on Linux.  The latest Cura for Linux works fine: however, there are no profiles for Dremel's 3D45 in them (there is a plugin for 3D20 that works great for that printer, but not so much for the 3D45).

Ghostkeeper, 2020-07-20:
@eduncan911 Your Cura version is rather old. Somewhere around 3.5 I'm guessing, based on those screenshots.

Your issue is different from what this topic is about. Please see https://github.com/Ultimaker/Cura/issues/6822 instead.

AndreVanKammen, 2020-10-12:
[image]
I keep having these barely readable fonts in Cura on my windows 10 PC throughout all versions. I'm working on a 4K tv, seems the font's are rendered to small so they fall in between pixels.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

Ywang1233, 2024-11-23:
I don't know. I have updated my pc os to win11, haven't seen it on win 11
yet.

GregValiant, 2024-11-23:
I'll close it then.

closed 2024-11-23 by GregValiant