Ultimaker/Cura #6791: Unable to slice one at a time - tuple index out of range

gergely-xyz, 2019-12-11:
**Application version**
4.3.0

**Platform**
Manjaro Linux x86_64
Kernel: 5.4.2-1-MANJARO 
Host: X570 AORUS ELITE -CF
CPU: AMD Ryzen 7 3700X 8- (16) @ 3.600GHz 
GPU: NVIDIA GeForce RTX 2060 Rev. A 

**Printer**
Creality3D Ender-3

**Reproduction steps**
Loaded a cable clip model from thingiverse 
https://www.thingiverse.com/thing:2960375
Duplicated 2 new copies of it
Set Print Sequence to "One at a Time"
Selected "Arrange all models" and confirmed that every model has enough place to be printed

**Actual results**
"Unable to slice" message is shown.

**Expected results**
I have sliced this project for the same printer with the same settings on Windows 10 before and it worked fine. 

**Project file**
https://we.tl/t-mytyM1REjo

**Log file**
I think this is the relevant part of the log. 
```
IndexError: tuple index out of range
2019-12-11 11:08:43,560 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [225]: Starting to slice...
2019-12-11 11:08:43,561 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [243]: Going to slice build plate [0]!
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [84]: Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x7efbf4e07d30> caused an exception
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [88]: Traceback (most recent call last):
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [88]:   File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 121, in run
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [88]:     job.run()
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [88]:   File "/usr/lib/cura/plugins/CuraEngineBackend/StartSliceJob.py", line 181, in run
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [88]:     for node in OneAtATimeIterator(self._scene.getRoot()):
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [88]:   File "/usr/lib/python3.8/site-packages/cura/OneAtATimeIterator.py", line 41, in __init__
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [88]:     super().__init__(scene_node)  # Call super to make multiple inheritance work.
2019-12-11 11:08:44,145 - ERROR - [Thread-10] UM.Logger.logException [88]:   File "/usr/lib/python3.8/site-packages/UM/Scene/Iterator/Iterator.py", line 18, in __init__
2019-12-11 11:08:44,146 - ERROR - [Thread-10] UM.Logger.logException [88]:     self._fillStack()
2019-12-11 11:08:44,146 - ERROR - [Thread-10] UM.Logger.logException [88]:   File "/usr/lib/python3.8/site-packages/cura/OneAtATimeIterator.py", line 121, in _fillStack
2019-12-11 11:08:44,146 - ERROR - [Thread-10] UM.Logger.logException [88]:     if self._checkForCollisions():
2019-12-11 11:08:44,146 - ERROR - [Thread-10] UM.Logger.logException [88]:   File "/usr/lib/python3.8/site-packages/cura/OneAtATimeIterator.py", line 96, in _checkForCollisions
2019-12-11 11:08:44,146 - ERROR - [Thread-10] UM.Logger.logException [88]:     if node_dict["bounding_box"].intersectsPolygon(other_node_dict["convex_hull"]):
2019-12-11 11:08:44,146 - ERROR - [Thread-10] UM.Logger.logException [88]:   File "/usr/lib/python3.8/site-packages/UM/Math/Polygon.py", line 184, in intersectsPolygon
2019-12-11 11:08:44,146 - ERROR - [Thread-10] UM.Logger.logException [88]:     ret_size = (polygon_intersection.bounds[2] - polygon_intersection.bounds[0],
2019-12-11 11:08:44,146 - ERROR - [Thread-10] UM.Logger.logException [88]: IndexError: tuple index out of range
2019-12-11 11:08:47,449 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
```

ninovanhooff, 2019-12-11:
Thanks for reporting this. I will discuss this with the team.

ninovanhooff, 2019-12-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/6771

closed 2019-12-11 by ninovanhooff

KidlandCode, 2020-05-20:
I have the same issue

arifdarmawan9, 2020-07-29:
I have the same issue. I think the files that I want to print is the problem. I wan to print these files https://www.thingiverse.com/thing:33732
And then I try to " repair" the files by open the files on Meshmixer and then I export it to STL. I open the resulted STL files using Cura,the problem is gone...:)