Ultimaker/Cura #6793: [master] crashing on startup

smartavionics, 2019-12-11:
**Application version**
master branch at 537424c058fc7074e9c793c27b2c5a39c0f4db36.

**Platform**
Linux.

**Printer**
SV01 based on CR10.

**Reproduction steps**
1. I started Cura.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Cura crashes and burns.

**Expected results**
Bliss.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

```
utdated. Its setting version is 10 but it should be 11.
2019-12-11 14:47:54,421 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [65]: Instance container creality_cr10_sv01_standard is o
utdated. Its setting version is 10 but it should be 11.
2019-12-11 14:47:54,424 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [65]: Instance container creality_cr10_sv01_standard is outdated. Its setting version is 10 but it should be 11.
2019-12-11 14:47:54,426 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [65]: Instance container creality_cr10_sv01_standard is outdated. Its setting version is 10 but it should be 11.
2019-12-11 14:47:54,427 - ERROR - [MainThread] UM.Logger.logException [85]: Exception: Error when loading container creality_cr10_sv01_standard: maximum recursion depth exceeded
2019-12-11 14:47:54,428 - ERROR - [MainThread] UM.Logger.logException [89]: Traceback (most recent call last):
2019-12-11 14:47:54,428 - ERROR - [MainThread] UM.Logger.logException [89]:   File "/opt/cura/lib/python3/dist-packages/UM/Settings/ContainerRegistry.py", line 183, in findContainers
2019-12-11 14:47:54,428 - ERROR - [MainThread] UM.Logger.logException [89]:     new_container = provider.loadContainer(metadata["id"])
2019-12-11 14:47:54,429 - ERROR - [MainThread] UM.Logger.logException [89]:   File "/opt/cura/lib/uranium/plugins/LocalContainerProvider/LocalContainerProvider.py", line 66, in loadContainer
2019-12-11 14:47:54,429 - ERROR - [MainThread] UM.Logger.logException [89]:     container = container_class(base_id)
2019-12-11 14:47:54,429 - ERROR - [MainThread] UM.Logger.logException [89]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 427, in new_new
2019-12-11 14:47:54,429 - ERROR - [MainThread] UM.Logger.logException [89]:     setattr(sub, key, Signal(type = value.getType()))
2019-12-11 14:47:54,430 - ERROR - [MainThread] UM.Logger.logException [89]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 149, in __init__
2019-12-11 14:47:54,430 - ERROR - [MainThread] UM.Logger.logException [89]:     if _recordSignalNames():
2019-12-11 14:47:54,430 - ERROR - [MainThread] UM.Logger.logException [89]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 75, in _recordSignalNames
2019-12-11 14:47:54,430 - ERROR - [MainThread] UM.Logger.logException [89]:     return FlameProfiler.enabled()
2019-12-11 14:47:54,431 - ERROR - [MainThread] UM.Logger.logException [89]:   File "/opt/cura/lib/python3/dist-packages/UM/FlameProfiler.py", line 23, in enabled
2019-12-11 14:47:54,431 - ERROR - [MainThread] UM.Logger.logException [89]:     return "URANIUM_FLAME_PROFILER" in os.environ
2019-12-11 14:47:54,431 - ERROR - [MainThread] UM.Logger.logException [89]:   File "/opt/cura/lib/python3.5/_collections_abc.py", line 603, in __contains__
2019-12-11 14:47:54,431 - ERROR - [MainThread] UM.Logger.logException [89]:     self[key]
2019-12-11 14:47:54,431 - ERROR - [MainThread] UM.Logger.logException [89]:   File "/opt/cura/lib/python3.5/os.py", line 722, in __getitem__
2019-12-11 14:47:54,432 - ERROR - [MainThread] UM.Logger.logException [89]:     value = self._data[self.encodekey(key)]
2019-12-11 14:47:54,432 - ERROR - [MainThread] UM.Logger.logException [89]: RecursionError: maximum recursion depth exceeded
2019-12-11 14:47:54,433 - ERROR - [MainThread] UM.Logger.logException [85]: Exception: Error when loading container SV01: list index out of range
```

I also get a lot of messages like these...

```
2019-12-11 14:47:47,637 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container flsun_qq_extruder_#2_pla,_0.2,_3+3+3,_@80,_200+50,_15%: {'intent_category': 'default', 'definition': 'fdmprinter', 'id': 'flsun_qq_extruder_#2_pla,_0.2,_3+3+3,_@80,_200+50,_15%', 'type': 'quality_changes', 'version': '4', 'name': 'pla, 0.2, 3+3+3, @80, 200+50, 15%', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'setting_version': '10', 'quality_type': 'draft', 'position': '0'}
2019-12-11 14:47:47,647 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container prusa_i3_mk2_extruder_0 #2_user: {'definition': 'prusa_i3_mk2', 'id': 'prusa_i3_mk2_extruder_0 #2_user', 'extruder': 'prusa_i3_mk2_extruder_0 #2', 'version': '4', 'name': 'prusa_i3_mk2_extruder_0 #2_user', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'setting_version': '10', 'type': 'user'}
2019-12-11 14:47:47,651 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container flsun_qq_extruder #2: {'enabled': 'True', 'machine': 'round-240', 'name': 'Extruder', 'type': 'extruder_train', 'version': '4', 'id': 'flsun_qq_extruder #2', 'container_type': <class 'UM.Settings.ContainerStack.ContainerStack'>, 'setting_version': '10', 'position': '0'}
2019-12-11 14:47:47,660 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container kossel_xl_kxl_artwork: {'intent_category': 'default', 'definition': 'fdmprinter', 'id': 'kossel_xl_kxl_artwork', 'type': 'quality_changes', 'version': '4', 'name': 'KXL_artwork', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'setting_version': '10', 'quality_type': 'normal'}
2019-12-11 14:47:47,673 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container Creality Ender-3 Pro_settings: {'definition': 'creality_ender3', 'id': 'Creality Ender-3 Pro_settings', 'version': '4', 'name': 'Creality Ender-3 Pro_settings', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'setting_version': '10', 'type': 'definition_changes'}
2019-12-11 14:47:47,675 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container custom_extruder_8_#3_m3d_promega_compound_june_05_abs_r: {'intent_category': 'default', 'definition': 'fdmprinter', 'id': 'custom_extruder_8_#3_m3d_promega_compound_june_05_abs_r', 'position': '7', 'version': '4', 'name': 'M3D Promega Compound_June_05_ABS_R', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'setting_version': '10', 'quality_type': 'normal', 'type': 'quality_changes'}
2019-12-11 14:47:47,727 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container ultimaker3_extended_extruder_left #3_user: {'definition': 'ultimaker3_extended', 'id': 'ultimaker3_extended_extruder_left #3_user', 'extruder': 'ultimaker3_extended_extruder_left #3', 'version': '4', 'name': 'ultimaker3_extended_extruder_left #3_user', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'setting_version': '10', 'type': 'user'}
```

Ghostkeeper, 2019-12-12:
So this happened because we updated the setting version for the upcoming Cura version to 11. You seem to have profiles, perhaps active ones, that are still on version 10.

I'm not getting a crash of Master when I start it, so in a way it's not reproducible. I also tried to change the setting version of my active quality profile to 10 but it didn't crash on that either, just displayed a warning in the log.

It should not crash on this. But I don't know how to reproduce the issue yet.

smartavionics, 2019-12-12:
If I zip up all the stuff under .local/share/cura/master and attach would that help find the problem?

smartavionics, 2019-12-12:
Here it is...

[cura-config-master-20191212.zip](https://github.com/Ultimaker/Cura/files/3954998/cura-config-master-20191212.zip)

Ghostkeeper, 2019-12-12:
Nop, no crash. Not even after switching to the SV01 printer.

smartavionics, 2019-12-12:
Could anything in my ~/.cache/cura tree cause a problem here?

Ghostkeeper, 2019-12-12:
Hmm, I wouldn't think so. We cache the definition files, but there is no setting version in those so that shouldn't be the problem. Of course it can't hurt to delete that folder. Cura will take a few seconds longer to start next time but that's not a problem.

smartavionics, 2019-12-12:
I think it's not loading the VersionUpgrade43to44 plugin although that plugin is in the Cura sources it's not actually installed?

smartavionics, 2019-12-12:
Sorry, I meant VersionUpgrade44to45!

smartavionics, 2019-12-12:
And that's because I hadn't run cmake after merging the recent upstream master.

Duh!

closed 2019-12-12 by smartavionics