Ultimaker/Cura #6794: Ajout de nouvelle machine

fantibus, 2019-12-11:
Bonjour , serai t’il possible d’ajouter une nouvelle marque à votre sélection de machine . 
Je suis sous zortrax ( slicer z-suite) je souhaiterai slicer avec cura car vraiment plus détailler . Le problème de zortrax c’est qu’il utilise leur propre gcode ( gcodex) . Cordialement dans l’attente de votre réponse merci

rburema, 2019-12-12:
Hi @fantibus 
For machine definitions and other things related to 3rd party machines we're wholly reliant on community contributions. (In other words: If it's not an Ultimaker machine, we expect people to make a pull-request.)
cheers, Remco.

fantibus, 2019-12-12:
Bonjour , je n’ai pas compris ce que vous voulez dire

rburema, 2019-12-12:
(Veuillez exuser la traduction de google.)

Des machines (autres que les machines Ultimaker) sont ajoutées à Cura par la communauté.

Quelqu'un doit créer une «définition de machine» ou un plugin.

Nous ne le faisons pas (pour les imprimantes tierces) nous-mêmes, mais sommes disposés à ajouter le code de programme qui nous a été donné à Cura.

fantibus, 2019-12-12:
Il faut donc que une personne développe le gcode pour zortrax ?

rburema, 2019-12-12:
Oui. Selon la quantité à changer, cela peut être dans une «définition de machine», un plugin, un script de post-traitement ou même le moteur qui génère le gcode lui-même.

rburema, 2019-12-12:
Si vous ajoutez simplement l'imprimante pour vous-même, vous pouvez également essayer si «FFF Printer» sous l'en-tête «Custom».

Il vous montrera une fenêtre où vous pouvez ajuster les saveurs (disponibles) de gcode, et changer le gcode de début et de fin pour l'imprimante dans son ensemble (et les extrudeuses, dans l'autre onglet).

fieldOfView, 2019-12-12:
Zortrax has a closed ecosystem. Their “zcode” is not documented, there is no open source implementation. So the likelyness of anybody writing a plugin for Cura to support zcode is extremely small.

The language used here in the Cura issue queue is English. If you need it to be translated, please use an automated translator or ask someone willing to translate for you.

fantibus, 2019-12-12:
Très bien merci , oui les gcode zortrax sont cryptés. Cordialement en
attendant que une personne ce penche sur le problème

rburema, 2019-12-12:
I'm sorry, but there doesn't seem to be much we can do. We'll have to defer this issue.

Ghostkeeper, 2019-12-12:
I think we have to close it even. We can't implement this for you. We don't have time, pay or knowledge to implement z-code.

Perhaps their own slicer can be called on to convert g-code to z-code. But we're not going to build that. This is something that Cura's plug-in ecosystem can bring you maybe, if someone is able to build that.

closed 2019-12-12 by Ghostkeeper