Ultimaker/Cura #6796: Print unattached circles

brybryguy, 2019-12-12:
When I try to print 2  unattached squares, they print fine. When I try to pint 2 unattached circles they will not print. I am triing to use my Ultimaker 2 as a line drawer and vinyl cutter.
[image]

Thank you

rburema, 2019-12-12:
Can you attach the model you're using? I can't really diagnose the problem from this description alone.

Ghostkeeper, 2019-12-12:
What happens when you load them into Cura, press slice and then press print? Could you fill out the issue template?

brybryguy, 2019-12-13:
I made the model using Spaceclaim 2019, it turns out that Spaceclaim must
have a bug in their software that saves Squares and circles in a way that
Cura does not see them the same.  When the stl file is viewed in preview
mode, the squares are hi lited red and the circles are not. I have
attached the file. When I made the same file in Sketchup 2019, it works
fine.

rburema, 2019-12-13:
I don't see the attached model on github. I don't think it works (on github) if you attach it to an email?

Ghostkeeper, 2019-12-13:
No, Github will ignore attachments to e-mail replies, probably because many e-mail clients will add previous e-mails as attachments and that sort of thing.

closed 2020-06-19 by no-response[bot]