Ultimaker/Cura #6797: Cura 4.4.0 - MDT/WDS Silent Installation

bwe1982, 2019-12-12:
**Application version**
Cura 4.4.0

**Platform**
Server 2019 / Windows 10

**Printer**
N/A

As part of my task sequence I am trying to install Cura silently. I have installed it silently using the following parameters Ultimaker_Cura-4.4.0-win64.exe /S with no issues. However my colleagues would like to open .OBJ files and that option is an optional extra that I cannot install unless I manually select that option. 

I was just wondering if there was any switches out there to select the options within the install process without any intervention.

Many thanks,

Ben

Ellecross, 2019-12-16:
For reference: CURA-7061 

@bwe1982  we discussed this in Cura team today and we created an internal ticket (new feature). :)

HeathHicks, 2020-06-23:
Is there a list of parameters for silent installation of additional drivers, etc?

Ghostkeeper, 2020-07-01:
For that you need to use the .msi installer rather than the .exe. We could not get those to work silently with the NSIS installer system.

HeathHicks, 2020-07-01:
> [EXTERNAL]: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Is there a download location for these?

Heath Hicks
IT Systems Administrator

Phone 574-389-3854 Cell 574-210-5972
heath.hicks@dometic.com

Dometic North America, 1120 N. Main St., 46514 Elkhart, IN United States

dometic.com<http://www.dometic.com>

[http://dometicpressroomfrontend.qbank.se/deployedFiles/338c47a2ce085fdc50b810927bfbd02d.png]<http://www.dometic.com>

[http://dometicpressroomfrontend.qbank.se/deployedFiles/a5d04859b4034100c185cad8220074a0.png]<http://www.facebook.com/dometic/> [http://dometicpressroomfrontend.qbank.se/deployedFiles/d9c545300165a1ff85a9f729d21bf97b.png] <http://www.instagram.com/dometic>  [http://dometicpressroomfrontend.qbank.se/deployedFiles/f5b9430e2561a2e2233863b3a844eb66.png] <http://www.youtube.com/dometicofficial>

NOTICE: This email and any attachments are for the sole use of the intended recipient(s) and may contain confidential information. Any unauthorized review, use, disclosure or distribution is prohibited. If you are not the intended recipient, please notify the sender by reply email and destroy the original message.

Ghostkeeper, 2020-07-01:
Yeah you can find them here: https://github.com/Ultimaker/Cura/releases

Ghostkeeper, 2020-07-06:
The MSI installer was provided partially for this feature. Fixed since Cura 4.5.

closed 2020-07-06 by Ghostkeeper