Ultimaker/Cura #6801: cura 4.4 ender 5 plus printer start G-code seems incorrect

NL-MGX, 2019-12-12:
i've installed cura 4.4 and it has a printer profile ready and waiting for the creality ender 5 plus. I've selected the profile and when printing immediately noticed the routine to clear the nozzle is off. The nozzle is moved through the area where the object is to be printed while cleaning the nozzle. This drags the nozzle through printed lines resulting in crap on the  nozzle or first layer adhesion failure, etc.

windows 10
cura 4.4
creality ender 5 plus 
working from memory card, no software or firmware mods on the printer.

With the previous version, the first two lines ended and then the nozzle moves diagonally towards the starting point of the actual print. This does not occur now.

current start g-code:
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

rburema, 2019-12-13:
Do you mean the 'prime blob'? If so, check the 'Enable Prime Blob' setting. It was defaulted to false in the base printer definition this year. If the profile doesn't override that, it'll be off.

Also: What version did you upgrade from? Between 4.1 and 4.2 we pulled in quite a few changes w.r.t. Creality.

Ghostkeeper, 2019-12-13:
The Enable Prime Blob setting is only applicable to Ultimaker 3 and newer Ultimaker printers. It outputs a `G280` command which was custom implemented by Ultimaker so Creality won't have any use for it and it should not display for them.

The start g-code for Creality Ender 5 was last edited by Trouch in Cura 4.2. The Cura developers can't edit it because we don't have an Ender 5 to test with. So we won't make any changes. But we'll accept changes from the Creality community if you've tested them. So please send us a better start g-code, preferably via a pull request, and we'll add it in!

closed 2019-12-13 by Ghostkeeper

chrisrafacz, 2021-04-03:
so what is the start and end gcodes for the ender 5plus bc im still getting the pause and blob and I can't seem to figure this out someone help!!! please!!
ive had the printer for 3 months and havn't gotten two decent prints out of it because of the pause and blog issue!!

rburema, 2021-04-09:
@chrisrafacz Go to the machine settings. (Either 'manage printers > machine settings' or 'preferences > printers > machine settings'.) You'll find the fields for start en end gcode there. (Also for the extruders, in another tab.)

cusz721, 2021-04-19:
I had the same issue with mine... the diagonal line is caused by an issue with these lines. i omitted them. since my cura adds a rift the print area, the prime line is unnecessary.

G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second lin