Ultimaker/Cura #6804: Ability to Add print information to G-Code file name when Saving.

mcolojr, 2019-12-13:
It's hard remembering things like what nozzle size or what material was used for an older G-Code file. So I usually manually add that information in the file name. It's a pain.

It would be nice there were some options to add print settings information to the .gcode file name when saving so we don't have to do it manually every time. Things like _Nozzle Size_, _Layer Height_, _Profile Name_, _Time Estimation_, _Material Used_, etc.

Example File Name:
_CE3_iPhoneXCase_.48LayerHeight_6mmNozzle_1h48m_PLA.gcode_

In **Preferences** > **General** 
We see there is an option to Add machine prefix to job name.
[image]
It would be nice if somehow there were more checkbox options for adding information to the .gcode file name. I feel these extra options would be appropriate under this section.

This would be a great feature for amateurs or professionals that like to stay more organized.

smartavionics, 2019-12-13:
Take a look at this plugin...

[image]

mcolojr, 2019-12-13:
@smartavionics Thanks for sharing! This plugin seems like a good solution although I'm having trouble figuring out if there is a way to add the objects name first. It automatically places it at the end.

smartavionics, 2019-12-13:
You need @fieldOfView to answer that. Cheers!

Ghostkeeper, 2019-12-16:
Duplicate of https://github.com/Ultimaker/Cura/issues/3753 (and solved by that plug-in that fieldOfView wrote too).

closed 2019-12-16 by Ghostkeeper

fieldOfView, 2019-12-16:
@mcolojr for the plugin only changes the _pre_fix that Cura already supports. Theoretically, the plugin could try replacing (more) parts of Cura to support a postfix, but this would increase the chance of the plugin breaking when Cura is updated.