Ultimaker/Cura #6805: [4.4.0] Rotation issues

benoitstjean, 2019-12-13:
**Application version**
4.4.0

**Platform**
Windows 7 64-bit 7.1 build 7601

**Printer**
Ulitmaker S5

**Reproduction steps**
References to 'printer' below is the printer representation in Cura and not the physical printer.
 
**1.** Check "Zoom toward mouse direction" and "Center camera when item is selected" from the general preferences.

**2.** Load a part on the printer as if it was going to be printed.

**3.** Press the 3D VIEW icon in the bottom-left corner (the first on the left, the 3D cube).

**4.** Right-click anywhere in the view to rotate the printer around.

--> The printer will rotate in the middle of the screen, probably with the pivot point centered in the xyz axis of the printer. This is expected and works exactly like in Solidworks.

**5.** To exhibit the problem, simply zoom-in by about 1/4 scroll wheel turns then repeat step 4 above. Now, the printer pivot point is totally off and the entire printer revolves around the screen making it impossible to rotate the view around the part. This occurs when the "Zoom toward mouse direction" and "Center camera when item is selected" are checked. However, the problem here is if the "Zoom toward mouse direction" option is not checked, the problem is not exhibited but then two other problems happen:

**5.A** Zooming-out too much makes that zoom-in no longer works and the printer remains "far" and the only way to zoom back-in is by clicking one of the bottom-left views and;

**5.B** Although the printer now rotates somewhat as expected, zooming-in will not zoom-in towards the pivot point therefore zooming-in too close will overshoot the part. Technically, this issue should be resolved by checking the "Zoom toward mouse direction" option but when doing so, that option causes the erratic rotation.... so is it that option that is broken?

**Screenshot(s)**
None. This must be tested by following the steps above.

**Actual results**
Described in steps 5, 5A and 5B above.

**Expected results**
View rotation should either always be referrenced to the xy center of the table if there's only one part (and by "table", it's where the 3D part sits -- the equivalent of the glass on the real printer) OR centered with the xyz center of the selected part when there are multiple parts. Again, the way Solidworks does it when rotating the view is how Cura should also work.

**Project file**
None.

**Log file**
None.

**Additional information**
None.

Ghostkeeper, 2019-12-16:
Unable to test this at the moment because of https://github.com/Ultimaker/Cura/issues/6775 :(

konskarm, 2019-12-16:
This happens because the anchor point around which the entire scene moves is static. This point is in the middle of your monitor.
[image]

When you press the 3D Cube the scene aligns the middle of the X-Y plane (but not the Z axis), i.e. the build plate, exactly on the X-Y of the anchor point and, as a result, moving the scene seems as if you are moving around the object.

When you zoom in while your mouse is randomly placed on the screen, the scene zooms in towards the mouse's direction, but the anchor point is still the center of your screen. 
[image]

Right clicking and moving around now looks like there is a weird behavior, but really the scene moves around the same absolute anchor point. **This is an expected behavior**. 

The best way for you to avoid this and always move the scene around the model, is to zoom in wherever you want and then re-click your model (while the "Center camera when item is selected" preference is set) so that your model is placed on the anchor point. Then, the scene will be rotating with the model as its center point.

benoitstjean, 2019-12-16:
Hi,

Thanks for the reply. Personally I think this is not natural and not very intuitive hence the reason I opened a ticket. I referenced Soliworks a few times because it works perfectly well when zooming and rotating.

Ben

rburema, 2019-12-17:
I can reproduce this. Re-clicking the model (or really, if it's already selected, unselecting it, _then_ reselecting it) isn't very intuitive. I'll discuss it with my colleagues to see if we can make a ticket or if this workaround is sufficient for now (considering our other work).

rburema, 2019-12-17:
We'd really like to have one of our UXers take a better look at the problem and the various resolutions available (like making center on object (which has to change the center) and zoom to mouse (which also does change the center) mutually exclusive).

She's on holiday right now, so this'll take a while.

benoitstjean, 2019-12-17:
Sure thing, no worries!

Thanks!

Ben

jimb15122, 2020-02-22:
While investigating view rotation issues, please consider fixing the tendency to _accidentally_ select a  right click context menu item when trying to right clicking and drag to change the viewing rotation.   (windows 10, and possibly others)  This happens to me several times per hour, and is very frustrating, because it is not usually obvious what action got selected.
Possible solutions - 
a) do not open context menu unless right click actually occurs on an object
b) Support use of left click+drag for rotation (possibly in addition to current behavior)

Ghostkeeper, 2020-02-27:
It should only open the context menu if you actually release the right mouse button. Perhaps your mouse clicker is getting a bit worn down?

jimb15122, 2020-02-27:
Thanks for the clarification on how it should work.  Yes, my clicker is no longer what it used to be, but who's is?  I will order a new mouse, and verify all is well.  Supporting left click drag would still be a very nice improvement.

mahtDFR, 2020-02-28:
If left click drag was supported, wouldn't it create the same problems by accidentally clicking on items on the build plate?
There are some tickets in our backlog to align the workflow with CAD software we might be rethinking how camera rotation works in the near future.

jimb15122, 2020-02-29:
The new mouse does work much better.  Sorry for the false alarm.

BlueLineWood, 2020-03-26:
I am having this same issue with the Camera Rotate except if i do the re-selecting the model work around it centres the model but zooms the screen all the way out and won't allow me to zoom in again.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

benoitstjean, 2024-11-22:
Hi,

Sorry, that was long ago. Yes you can go ahead and close.

Thanks!

Ben

closed 2024-11-22 by GregValiant