Ultimaker/Cura #6806: [4.4.0] Profiles "Not Supported" and "unable to slice" on several 3D printers with variants

DragonJe, 2019-12-13:
**Application version**
4.4.0

**Platform**
Mac OSX, Windows 10

**Printer**
IMADE3D JellyBox 2
NWA3D A31
FABtotum Personal Fabricator
HMS434
Deltacomb 3D
possibly more

**Reproduction steps**
1. I Added one of the printers above in Cura 4.4 (and upgrade Cura version to 4.4 on Windows and experienced the same problem)
2. I tried changing to different materials and still got "not supported" message for the profile name

**Screenshot(s)**
[image]

[image]

[image]

[image]

[image]

[image]

**Actual results**
Cura "unable to slice" for that 3D printer. (see screenshot above)

**Expected results**
The profiles should work correctly, as they did in the Cura 4.3 versions.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

https://www.dropbox.com/s/j57ffuv6gjnqzdr/A31.3mf?dl=0)

https://www.dropbox.com/s/yj1tepny8l0uobh/JellyBOX%202.3mf?dl=0

**Log file**
None

**Additional information**
I looked at almost every machine (even ones I don't own), and it's not a problem for every variant; for instance Ultimaker, TiZYX, or Creality machine variants still work fine.

The printer profiles listed worked before, so I think it could be something with the way variants are processed Cura 4.4.0. If there is anything I can fix with a pull request, let me know and I'd love to help get these printer profiles working again!

Ghostkeeper, 2019-12-16:
Looks like there are some mistakes in the profiles there. The causes of each of these are different.

* Jellybox 2: Caused by the Prime Speed being set to the Retraction Speed - 30 even though the default is 25, so unless the retraction speed is overwritten it'll end up with a negative speed: https://github.com/Ultimaker/Cura/blob/8b9e5c51548a261b322560de30944a5ab1fb6ecb/resources/quality/imade3d_jellybox_2/jb2_global_coarse.inst.cfg#L38 I changed it to have a minimum of 5mm/s, so that should be okay.
* NWA3D A31 doesn't have material-specific quality profiles even though the definition says that it should have. For this reason the material is marked as "not supported"; there is no profile for it. Not sure how to fix this one. We sort of need to test with the NWA3D A31 printer to create a profiles but I don't want to delete the global profiles that they do have either.
* FABtotum has profiles but they are not specific to the nozzle size. They added nozzle sizes later but didn't update their profiles.
* HMS434 doesn't show any problems to me. They support Generic PLA.
* Deltacomb has the same problem as FABtotum.

There are probably many printers with profile issues. Cura's profile system is very complex. When we merge something we do (or should) test the base case but we don't try every combination of profiles either so there will almost always be cases where the profiles are incomplete or just wrong. That said, some of these profile mistakes here make me wonder whether the author of those profiles even tested them and whether we tested them when we merged them...

Ghostkeeper, 2019-12-16:
I'll close this issue. I've fixed what I could but the rest is up to the maintainers of those profiles to fix. We cannot help you further.

closed 2019-12-16 by Ghostkeeper

DragonJe, 2019-12-16:
Ok, that makes sense. HMS434, only has one selectable quality, but there are three in their qualities folder. That's why I added it to this list, but I should have been more clear with why.

I'll go through and make pull requests for what I can to help them function. Thanks for the insight, @Ghostkeeper!