Ultimaker/Cura #6809: Import Project ignore setting option

LostKatana, 2019-12-14:
**Is your feature request related to a problem? Please describe.**
When importing a project the summary dialog has two options:
* Update printer/profile
* Create new printer/profile
When now importing an older Project when some settings in the printer have changed you always need to adjust the option to create a new printer you might delete afterwards as you already have tuned the setting since last opening of the project.

**Describe the solution you'd like**
I'd like to have an option to ignore settings as I'm saving only some print settings and also get some 3mf files shared by friends using other printers or firmware on same printer.

**Describe alternatives you've considered**
Only importing the models an setting the print settings manually can be a hazzle depending on what has changed.

**Affected users and/or printers**
I think everyone sharing 3mf files can benefit from not having to import a printer not used

**Additional context*
[image]
*

nallath, 2019-12-16:
If you don't want to change the settings, how is it different from just loading the models?

LostKatana, 2019-12-16:
I'd sometimes only like to import the profiles without having to import the printer as well. This e.g happens when I use special settings for a print.

Maybe we can have the profile temporarily set?

Ghostkeeper, 2019-12-16:
You can import the project, then delete the printer.