Ultimaker/Cura #6811: Preventing Elephant foot ignores first line

FlailAway, 2019-12-15:
Man, finding this drove me nuts for quite a while. It is pretty obvious now, but I have all profiles set for reducing Elephant Foot by defining Initial layer Horizontal Expansion = -0.3mm

**Application version**
4.4.0

**Platform**
Mint 19.2+Cinnamon

**Printer**
Any

**Reproduction steps**
Have a wall on the build plate only 0.5mm total width
Set Initial layer Horizontal Expansion to -0.3

Slice and the entire wall is missing from the first layer.

**Expected results**
if the Initial layer Horizontal Expansion totals less than the wall width then ignore the Initial layer Horizontal Expansion setting. It is far better to have an Elephant foot _plus_ a wall, than no wall on the first layer.

Ghostkeeper, 2019-12-16:
Currently this is expected behaviour so I'm changing it into a feature request. You're telling Cura to remove 0.3mm from all sides of the polygon. So it does that and your wall of 0.5mm gets 0.3mm removed from both sides, resulting in a wall of -0.1mm which gets deleted completely.

Ghostkeeper, 2019-12-16:
The problem with this is that a polygon's width is ill-defined. For instance, what's the width of this shape here?

[image]

It's about 2 units at the minimum. But that "wall" graduates to a width of about 4 units. If you have a line width of 3 units, should it perform an offset of this polygon or not?

What you're really asking for is to make it perform an offset of only the parts of the polygon that are thinner than the line width. This can theoretically be done. What you'd want is something like this:

[image]

The algorithm for that is pretty complex in itself. It involves getting a skeleton for the polygon which we don't currently have a library for. This skeleton needs to be offset with half the line width (something we do have an algorithm for in our source) and then unioned with the original shape.

However this is ignoring the other parts of the polygon that are also thinner than the line width that you'd also want to offset:

[image]

And that's not the print results you wanted. You probably wanted those lines to be sharp, not blocky. It'll introduce all sorts of extra vertices at every outside corner too. So you will need to do some sort of filtering. It's getting somewhat complex there. Is this approximately what you meant?

I've written an OpenSCAD script once to modify an object to have a certain minimum width. You can download it here: https://github.com/Ghostkeeper/SlicerTestModels/blob/master/my/make_thin_pieces_printable.scad

It takes a long time to process, but still less long as the time it takes for us to implement something like this.

smartavionics, 2019-12-16:
> The algorithm for that is pretty complex in itself. It involves getting a skeleton for the polygon which we don't currently have a library for. This skeleton needs to be offset with half the line width (something we do have an algorithm for in our source) and then unioned with the original shape.

This is, essentially, how my thin wall/gap filling works. I do think that in the long term, Cura's whole wall printing strategy needs to work along the lines of determining the walls' centre lines and then hanging one or more extrusions off of those (depending on the required wall width).

Ghostkeeper, 2019-12-16:
We have a research project going on to fit lines within a shape that uses the skeleton, but it's more of a long-term thing.

FlailAway, 2019-12-16:
> so I'm changing it into a feature request.
I beg to differ, leaving out an entire layer is not a Feature!

> You're telling Cura to remove 0.3mm from all sides of the polygon.
I fully understand what it is doing, I didn't think I needed to explain that in detail to you folk, but I did add in the Op, that it is better to have a wall than honour the set-back.

closed 2019-12-16 by FlailAway

Ghostkeeper, 2019-12-17:
> I beg to differ, leaving out an entire layer is not a Feature!

Then again, you weren't asking for an entire layer being left out. You were asking for thin parts of a layer to not be left out, which is a feature.