Ultimaker/Cura #6818: 4.4 Failed importing of old profile from 4.2

Jakobud, 2019-12-17:
**Application version**
4.4.0

**Platform**
Windows 7 x64. Radeon RX 580

**Printer**
Ender 3

**Reproduction steps**
Import the following profile: 
https://www.3dprintedtabletop.com/download/cura-siepie-cr10-ender-3-small-minis-pla-base-1-0/

**Screenshot(s)**
No screenshot but here is the error text:

```
Failed to import profile from D:/Downloads/Siepie-CR10-Ender-3-Small-Minis-PLA-Base-1.0 (1).curaprofile: Could not find a quality type normal for the current configuration.
```

**Actual results**
The profile is not imported

**Expected results**
The profile should be properly imported even though its for an older version of Cura

**Project file**
N/A

**Log file**
```
2019-12-17 09:52:43,967 - DEBUG - [MainThread] cura.Settings.CuraContainerRegistry.importProfile [175]: Attempting to import profile D:/Downloads/Siepie-CR10-Ender-3-Small-Minis-PLA-Base-1.0 (1).curaprofile
```

**Additional information**
This profile is for an older version of Cura, I think maybe 4.2.

Ghostkeeper, 2019-12-18:
This is expected behaviour. Those profiles were exported by Cura 3.5. The CR10 printer had a quality level "normal" back then. But the profiles have changed, so that quality level is no longer present. Cura doesn't know which quality profile to select for your profile since it doesn't exist any more.

I've modified that profile for you. This one should load. I've mapped it to a different quality level that does exist. This will get some of the settings back but it will 90% be different settings so don't expect the same print to roll out: [Siepie-CR10-Ender-3-Small-Minis-PLA-Base-1.0.zip](https://github.com/Ultimaker/Cura/files/3977801/Siepie-CR10-Ender-3-Small-Minis-PLA-Base-1.0.zip)

closed 2019-12-18 by Ghostkeeper

Count-zero-78, 2020-01-19:
@ Ghostkeeper Big Thx to you. Works fine also for me! i think a lot of people are strugglng with that!