Ultimaker/Cura #6819: [4.4] Object not printed if "Print as Support" selected with Tree Supports enabled

Silver-Streak, 2019-12-18:
**Application version**
4.4

**Platform**
Windows 10

**Printer**
Anycubic i3 Mega

**Reproduction steps**
1. Loaded this STL https://www.thingiverse.com/thing:1475453
2. Selected Tree Supports
3. Tried adding a square box to add more support to the knife area.
4. Clicked the box model.
5. Clicked Per Model Settings.
6. Selected Print as Support

**Screenshot(s)**

Square box model from Tinkercad, added to print area, then per model setting set to "Print as Support"
[image]

What it looks like with Tree Supports disabled:
[image]

What it looks like with Tree Supports enabled: 
[image]

What it looks like with Normal Supports enabled: 
[image]

**Actual results**
Only the first layer of the support is printed, then it is ignored.

**Expected results**
The custom support to be printed along with the Tree Support. I had used this a ton of times before in the past (4.2 I believe) with no issue.

**Project file**
Project file: 

[AI3M_Pyramid_head_V1.zip](https://github.com/Ultimaker/Cura/files/3976095/AI3M_Pyramid_head_V1.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/3976098/cura.log)

Ghostkeeper, 2019-12-18:
Yup, this reproduces.

The problem seems to be that the Tree Support Wall Line Count setting is not being applied to these shapes. The reason for that is that this is only applied when drawing the circular branches of the trees. These support areas are generated separately from the model outline.

For now I have a workaround for you:
* Enable the normal "Generate Support" setting.
* Change the "Support Wall Line Count" setting to 1.
* Disable the normal "Generate Support" setting again.

As a fix, I think we should merge the Support Wall Line Count setting with the Tree Support Wall Line Count.

Silver-Streak, 2019-12-22:
Just wanting to chime in that your workaround fixes the issue for me (although it seems like Support X/Y distance acts odd bit odd with that enabled, I didn't have enough time to test more to see the specifics or if it was just me overanalyzing).

Glad to hear I'm not crazy, and thanks for looking into this.

Ghostkeeper, 2019-12-23:
Fixed this for the next release.

closed 2019-12-23 by Ghostkeeper