Ultimaker/Cura #6822: Cura on Virtualbox - Text non readable and upsidedown

Pomaroli, 2019-12-18:
Application Version
4.3 / 4.4

Platform
Virtualbox 6.1 / Windwos 10 Desktop 
Host Windwos 10 

Printer
Custom

Steps to Reproduce
Open / start Cura

Actual Results
Cura shows up but everthing is upside down non readable

Expected results
User interface showing.

Additional Information
Probably wrong SVGA emulator

screenshot_cura
[image]

nallath, 2019-12-18:
We're going to need a lot more info to debug this, but I also doubt if there is something we can do to fix this (Since only parts of the UI have this issue, I expect that this is either an issue with the way your virtual machine works or with Qt)

Pomaroli, 2019-12-18:
Let me know what informations you need

Ghostkeeper, 2019-12-19:
I have no idea how to debug a thing like this except to search for similar problems online. And I found one here: https://forums.virtualbox.org/viewtopic.php?f=2&t=92944#p448138
One suggested fix is to allow blacklisted drivers in your virtual machine. Worth a try?

Could be that your OpenGL driver in that VM is using the convention of Y going down the screen instead of up.

Pomaroli, 2019-12-20:
I think this is not possible in virtualbox only in vmware with linux i guess.
Probably it is an opengl issue related to virtualbox as it was introduce recently, or any other suggestions ?
Strangewise i could go thru the wizard by clicking were the buttons "should" be and below how it looks like in cura. 

[image]

stendarr, 2020-04-11:
@Pomaroli You probably need 3D acceleration for this but just out of curiosity, what happens when you disable it in the virtualbox settings? I had a similar issue with text rendering upside down but still going left to right which disappeared after disabling 3D and 2D Video Acceleration in the display settings.

I think this might be related: https://www.virtualbox.org/ticket/18914

I had this issue on a laptop with Intel's integrated graphics (620 HD, i7-7600U).

Pomaroli, 2020-04-11:
i tried to disable 3D acceleration but when starting cura i get the error "could not probe Opengl. This programm requires Opengl 2.0 or higher....." didi you get some similar ?

I have also a laptop with I7 8650U and HD 620

eduncan911, 2020-07-20:
FYI, I'm having the exact identical issues:

* Fonts exactly as OP picture
* Build plate upside down, just like the pic above!

Nice to see I am not crazy.  :)

Hypervisor: VirtualBox (latest and greatest)
Host: Arch Linux + Gnome-i3
VM: fresh install of Windows 10 1904, and patched 

Cura is a rather old version, as that's all Dremel offers in their Dremel-flavored Cura edition.  I can't get to any About screen to know, but I think it's around 3.5-ish judging by the interface.  I use Cura 4.6 for my non-Dremel printers.

For now, I just dug in and found the json files for Dremel printer's configuration, opened them up, and translated the settings to Cura 4.6.x AppImage running on Linux native and it works fine.  Also translated the settings for PrusaSlicer appimage which works fine as well.

So the root issue is running Cura within VirtualBox.  And judging by the OP, it doesn't matter if the host is a Windows 10 or Linux OS.

enurseitov, 2020-08-10:
Same issue

casiosmu, 2020-09-18:
I can confirm the issue:
VirtualBox 6.1.4
Host: OpenSuse Tumbleweed
Guest: Win10 1909 x64

Whats interesting:
The Creality Slicer v1.2.3 thats based on Cura works without problems.
Do they use only the engine and have their complete separate GUI?

Ghostkeeper, 2020-09-21:
Creality Slicer is based off a 5 year old version of Cura. We've indeed reworked the GUI since then. We're now using the Qt framework for GUI which uses OpenGL to render many things. It seems that [there is a bug in VirtualBox or OpenGL that causes things to render upside down](https://forums.virtualbox.org/viewtopic.php?t=92944&start=15) when using the VBox-SVGA graphics controller.

ulfur, 2022-05-03:
Still present:

VirtualBox 6.1
Host: Ubuntu 20.04
Guest: Win10

fieldOfView, 2022-05-03:
Still an issue that should be fixed in VirtualBox

Dire-Pants, 2022-08-10:
VirtualBox 6.1.36 r152435 (Qt5.6.3)
Host: Max OS X 10.13.6 (Early 2011 MBP Intel HD Graphics 3000 512MB)
Guest: Win10 Education 21H2

- UI text not upside down but build plate is.
- Can run up to 4.12.0 on High Sierra with no issues but admittedly I was trying to force the use of Cura 5.1.0 in a VirtualBox environment on dated hardware
- Was able to replicate the build plate/volume being upside down with no clear way to adjust this
- Despite the quirk, slicing does still work, I just need to stand on my head to make sure everything looks good :stuck_out_tongue_closed_eyes: