Ultimaker/Cura #6823: [4.4] Generic Nozzle Angle of 45* in all Printer Definitions

kazooless, 2019-12-18:
**Application version**
(The version of the application this issue occurs with.)
4.4

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
ALL

**Printer**
(Which printer was selected in Cura?)
ALL / Ender3 / DiY-CoreXY

**Reproduction steps**
1. (Something you did.) Created new printer definition for my DIY-CoreXY machine which is using a genuine E3D v6 nozzle. Knowing the geometry of the nozzle is different than the Creality nozzle used in my other printer, I looked for another machine definition to use as a template. I found that all printer definitions either have "machine_nozzle_expansion_angle" set to 45, or they are based on the fdmprinter.def which is also 45. Other settings, such as "machine_nozzle_head_distance" also differ from the published geometry.

2. (Something you did next.)
Then I found the Creality CAD drawing for their nozzle and it is different than the generic measurements as well.

In addition, some of the descriptions for the nozzle measurements are ambiguous. I will ask for clarification at the end of this report.

** Screenshots**
[image]
[image]

**Additional information**
(Extra information relevant to the issue.)

Here is the E3D CAD drawing: https://wiki.e3d-online.com/images/3/3a/V6-NOZZLE-ALL.pdf
Here is the Creality CAD drawing: https://github.com/Creality3DPrinting/Ender-3/blob/master/Ender-3%20Mechanical/PDF/nozzle.pdf

**Questions for the Developers**

Is the "machine_nozzle_expansion_angle" referring to the outside angle of the metal, or the inside angle of the bore where the filament melts? Both of these nozzles do not have the outside parallel with the inside angle.

Is the "machine_nozzle_head_distance" what I pointed out in the screenshots (2mm for E3D, 4mm for Creality)? The description as it is given now says: "The height difference between the tip of the nozzle and the lowest part of the print head." What is "the lowest part of the print head."? Seems to me the tip itself is the lowest part. There is the bottom of the heat block itself too.

For "machine_heat_zone_length," the description says "The distance from the tip of the nozzle in which heat from the nozzle is transferred to the filament." Using the E3D PDF linked above, is this just the entire length of the nozzle, 10.5mm? Or is it maybe 6mm, the length of the thread? Or is it from the tip to the top of the heat block itself?

Ghostkeeper, 2019-12-19:
> I found that all printer definitions either have "machine_nozzle_expansion_angle" set to 45, or they are based on the fdmprinter.def which is also 45.

Yeah, good point. Those overrides are useless because they override them with the same value. I removed them.

> Is the "machine_nozzle_expansion_angle" referring to the outside angle of the metal, or the inside angle of the bore where the filament melts? Both of these nozzles do not have the outside parallel with the inside angle.

It refers to the outside angle of the metal. The E3D nozzle drawing doesn't list it. From the Creality drawing you can infer that it should be `90 - 74/2 = 53` degrees. This setting is currently only used during Wire Printing, [here](https://github.com/Ultimaker/CuraEngine/blob/a1f55b023958ebc8494c51c810b379b66e4995d5/src/Weaver.cpp#L408). There it's used to determine the spacing between the vertical edges of the wire frame.

> Is the "machine_nozzle_head_distance" what I pointed out in the screenshots (2mm for E3D, 4mm for Creality)?

In technical terms, it's the height up to which the expansion angle is still indicative of where the nozzle can still move without colliding. So in the E3D nozzle that height will indeed be 2mm as you indicated. In the Creality nozzle you can still keep going at that 53 degree angle for a while longer without hitting anything (it's not in the drawing). At some point you'll hit something else than the nozzle. This is again only used for wire printing to determine where the nozzle can move to print the wires.

> What is "the lowest part of the print head."? Seems to me the tip itself is the lowest part. There is the bottom of the heat block itself too.

Yeah so it's the first thing that you would hit if you were to go up the nozzle except the part that falls within that nozzle expansion angle. So if there's a heater block there that heats up the nozzle, that's the first thing you hit. If there is not, then you can go quite a bit further and you'll give Wire Printing a bit more freedom of movement.

> For "machine_heat_zone_length," the description says "The distance from the tip of the nozzle in which heat from the nozzle is transferred to the filament." Using the E3D PDF linked above, is this [...]

The "heat zone" is the length along the filament that the printer uses to heat up the material. Generally it's the part of the nozzle in contact with the heating element. Many (all?) nozzles will have a heat break to separate the hot part from the sensitive electronics in the print head. Above the heat break it'll be cold and below it it'll be at printing temperature. Cura uses this setting as default the extruder switch retraction distance and the parking retraction distance (after the extruder is done with its last layer but other extruders still continue). So essentially what Cura now expects is that beyond the heat zone the material will be below the glass transition temperature so that the material doesn't degrade even if it stays there for hours.

Ghostkeeper, 2019-12-19:
> (How do you add different labels?)

I think only the maintainers can do that. Of course if it doesn't make sense to fill in the bug template then we won't force you to. Human touch :)

Ghostkeeper, 2019-12-19:
I've removed all of the extraneous nozzle angles. And the questions are answered, so I'll consider this fixed and close the issue. But if you have more questions about this topic you can still continue discussion.

closed 2019-12-19 by Ghostkeeper