Ultimaker/Cura #6827: cura 4.4.0 wont start on Win10pro and Nvidea GTX - fatal error

Archibald-Harry-Tuttle, 2019-12-19:
**Application version**
Cura 4.4.0 wont start after installation- Cura 4.3.0 works fine
At start a "fatal error has occurred in cura" is shown and application is stopped 

**Platform**
Intel I7-8700
16GB RAM
Nvidea GTX1060i 3GB RRAM

**### Remark: in "ultimaker forum" you can find (searching for "fatal error") exakt same bug/error from other user, using also GTX 1070 and 1080 grafic boards - same error**

**Printer**
UM0
Ultimaker 2 with Olsson Block
UM3

**Reproduction steps**
Cura 4.4.0 installed, de-installed many times
Cura 4.4.0 beta - same error

Cura 4.3.0 re-installed -> working fine

**Screenshot(s)**
none
**Actual results**
none

**Expected results**
App will crash at start with "fatal error" and stopping

**Project file**
none

**Log file**
Thread 0x00003250 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000045d0 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.4\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000020c4 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002850 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002fdc (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.4\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000044e0 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.4\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
[... 121 lines cut ...]
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Event.py", line 189 in call
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 428 in event
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 1003 in event
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 341 in exec_
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 804 in run
  File "X:\4.4-exe\build\inst\bin\cura_app.py", line 180 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

**Additional information**
all user facing this fatal error seem to have Nvidea GTX graphic boards

Would appreciate if someone can help or give a hint how to fix

thank you,
regards
Michael

Ghostkeeper, 2019-12-20:
Hmm, at home I have an NVIDIA GTX2070, running through Windows 8, but haven't seen this problem (yet).

Unfortunately that crash log doesn't show anything of interest as it's tracing a log to the crash dialogue itself? It almost looks like Cura got a crash and then when showing the crash dialogue it logs a crash again.

When exactly are you seeing this crash? Immediately after starting Cura, or do you see a splash screen (and what text on it?) or do you get a Cura main window and then a crash?

Archibald-Harry-Tuttle, 2019-12-21:
Hello Ghostkeeper

 

Thank you thirst for responding.

Never had any problem until cura 4.4.0 (and also beta) – cura 4.3.0 still running without any issue

 

The start of Cura is as follow:

White screen

Followed by Crash Report

Half second later the starting screen of cura is shown, but not reachable by mouse.

Ist showing my UM2 with ist setting on right side.

 

When I close the Crash Report, also Cura is closing

 

Regards

Michael

Ghostkeeper, 2019-12-23:
Ah, so it's getting through the splash start-up sequence then. Good sign.

Maybe you can post a log file as well? It's located in `C:\Users\`<username>`\AppData\Roaming\cura\4.4\cura.log`.

Archibald-Harry-Tuttle, 2019-12-23:
Hello Ghostkeeper

 

Thank you!

 

Here ist the requested .log file:

 

Kind regards

Michael

 

Maybe you can post a log file as well? It's located in C:\Users\\AppData\Roaming\cura\4.4\cura.log.

Ghostkeeper, 2019-12-24:
Please note that attachments don't work when sent via an e-mail reply to Github. Your reply contains no log file.

Archibald-Harry-Tuttle, 2020-01-12:
Hello Ghostkeeper

 

Thank you for your support.

My problem was fixed by following these actions, provided by a member of cura forum.

 

https://community.ultimaker.com/topic/30276-cura-44-fatal-error/?tab=comments#comment-251257

 

Kind Regards

Michael

closed 2020-06-19 by no-response[bot]