Ultimaker/Cura #6829: Scale

Sadesign823, 2019-12-19:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2019-12-20:
I can't reproduce this problem. This is a screenshot of the scale tool panel showing the absolute size and the scale percentages.

[image]

It doesn't look like there is anything mixed up. The scale is in the scale textboxes and the percentage is in the percentage textboxes.

Can you provide us with:
* Reproduction steps
* Expected behaviour
* Observed behaviour (and how it's different from expected behaviour)

We have an issue template that asks for these things but you've neglected that. It would help to solve this problem!

GooodFr, 2019-12-30:
Hello,

I have the same problem with 4.4.0 & 4.4.1 (never tried before) on PC

To reproduce the problem : 
 _ Import a file (I like so much the drag and drop 👍 )
 _ Rotate your object to 90°
 _ Unselect Uniform Scaling
 _ Change axis values
1) Some percent are not in front of the correct axis
2) If you correct the percent auto-modified, the other axis is modified ! ))

Regards ;)

Ghostkeeper, 2019-12-31:
Ah, yeah I understand then. This is a duplicate of https://github.com/Ultimaker/Cura/issues/4136. The 20th duplicate in fact :tada: We understand it's a big problem, but it's a very hairy problem to tackle. We've concluded now that the interface needs to be changed to make the scaling by percentage apply only to the current state and then reset to 100% after the scaling is completed.

closed 2019-12-31 by Ghostkeeper