Ultimaker/Cura #6830: Cura 4.4 issue with login at startup_cura wizard each time

Jasontwd, 2019-12-20:
Cura 4.4 issue with login at startup, why does it defaut back to the setup wizard.

Each day I start Cura, application does not remember the login and takes me thru the setup wizard.

From there I have to login to the main website, go thru the setup wizard and set custom printer

then re load a restore from the backups of the previous day.

Re open cura, and delete new profile.

Then I get the config error to deal with each time.

Didnt happen in previous versions of cura.
No Logs available, Just need login on start not wizard.
 
Can someone please add a login screen to the appliaction on startup?????????? To avoid the wizard.
See post in community here:
[https://community.ultimaker.com/topic/30560-cura-44-issue-with-login-at-startup_cura-wizard-each-time/?tab=comments#comment-253194](url)

Anyone else having this problem??

cheers

Jason

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2019-12-20:
Hi @Jasontwd 
Could you please share the log file(s)? They' should be in `C:\Users\`_username_`\AppData\Roaming\cura\4.4`

Jasontwd, 2019-12-20:
Thanks for that @rburema,
Attached are the logs from the directory.
cheers J
[cura.log](https://github.com/Ultimaker/Cura/files/3989488/cura.log)

Jasontwd, 2019-12-20:
Thanks Remco,

copy attached, I'll upload on the thread too on git
cheers

Ghostkeeper, 2019-12-23:
Several observances in the log here:

* Getting a configuration error about the "Cube Pro" custom printer. Something got corrupt there. It seems that the extruder stack user settings profile got deleted? Did you meddle with the files there?
* Same with the user settings for the Ultimaker S5 printer that gets added in the last run. Perhaps the entire `/user` folder got deleted?
* The USB printing plug-in is getting a time-out when trying to detect printer devices. Probably unrelated.
* At some point you're getting a division by zero error because layer height is zero. Probably unrelated as well, you may have been changing the layer height in the interface.

I think your safest bet would be to close down Cura, then go into that configuration folder and delete everything in `machine_instances` and `extruders` related to "custom" (the Cube Pro printer) and "ultimaker_s5". Those two printers seem to have gotten broken because something deleted the `user` files. Then just to be safe, also remove them in `definition_changes` and `user` if there is anything left there. Next time you start Cura you'll be asked to add a new printer and that new printer should be fine again.

The reason why you keep going back to the add printer wizard is most likely because you keep loading the broken back-up. When the current printer can't be loaded, Cura will ask you to add a new printer.

closed 2020-06-19 by no-response[bot]